corneliusmunz/legoino #2: TrainHub Does not seem to work

ebartz, 2019-08-22:
Hi There,

when I flash the train_esp example to my esp32, it is able to connect to the hub and also change the LED color. But it seems like it is not sending the right command for controlling the motor speed.

I tried both of the ports. Maybe this does not work for me, since lego installed an update to my hub when I used the PoweredUP App on iOS?

Thanks for helping me out :)

ebartz, 2019-08-22:
I don't really know why this is the case, but when I change the byte sequence I am able to control the train speed:

https://github.com/ebartz/legoino/commit/340cd046b34cf732f845e13bdb6d2634071e0612

I have looked at the node-poweredup project did here:
https://github.com/nathankellenicki/node-poweredup/blob/master/src/puphub.ts#L129

Maybe this is due to firmware updates on the hub?

docteurzoidberg, 2019-10-06:
@ebartz check my issue about star wars boot hub it may helps !

corneliusmunz, 2019-10-06:
@ebartz Thanks for reporting that issue. I think the powered up Hub update has changed the commands. I have checked it on my train hubs and your changes works. The AB Port does not work even with the Port changes proposed by @docteurzoidberg. I think for the train hubs you can only access specifically Port A or Port B and not both at the same time. I will change the library with your input and will release a new version. @ebartz, @docteurzoidberg  Again hands up for your feedback and your patience ;-)

corneliusmunz, 2019-10-06:
@ebartz Fixed with commit: https://github.com/corneliusmunz/legoino/commit/ae75825918beacf1338814acbf75eb44d5a72bb8

closed 2019-10-06 by corneliusmunz