corneliusmunz/legoino #3: Boost hub motors not seems to work with latest firmware

docteurzoidberg, 2019-10-05:
Tested with the lego star wars boost r2d2 droid with latest firmware:

I managed to get connected to hub , set the leds, but no motors actions seems to work.
Shutdown worked fine also

docteurzoidberg, 2019-10-05:
i made moto spinning by doing: 

myBoostHub.setMotorSpeed(BoostHub::Port::C , 25);
delay(2000);
myBoostHub.setMotorSpeed(BoostHub::Port::C, 0);

but it's not the C channel motor wich spins*

docteurzoidberg, 2019-10-05:
class BoostHub : public Lpf2Hub
{
public:
    //Port definitions specific to Boost Hubs
  enum Port
  {
    A = 0x00,
    B = 0x01,
    AB = 0x10,
    C = 0x02,
    D = 0x03,
    TILT = 0x3A,
    CURRENT = 0x3B,
    VOLTAGE = 0x3C
  };

corneliusmunz, 2019-10-06:
@docteurzoidberg Thanks for reporting that issue. Unfortunately i have not done so much in the last months but definitely lego has changed the port assignments in the hub updates. In the official documentation this is not commented but in the superb node project from @nathankellenicki i found also the port changes too (https://github.com/nathankellenicki/node-poweredup/commit/7b61b6981a0b3a5f6ae8af8559e6560ac5962fc1#diff-68f9133fb0b523de7ff1ecbbd0f48229) I have also checked it with my boost hub. I will merge your pull request!

corneliusmunz, 2019-10-06:
@docteurzoidberg: Fixed with the following release: https://github.com/corneliusmunz/legoino/releases/tag/0.3.0

closed 2019-10-06 by corneliusmunz