corneliusmunz/legoino #5: The motor does not rotate

boomboomsatellites, 2019-11-15:
Hi !,
I'm trying to move the sample, but the motor doesn't move.

The LED seems to work.

Please tell us the Firmware version of Boost that you are testing.

This uses the latest version 1.0.00.022.

Thank you for your answer.

corneliusmunz, 2019-11-15:
Hi @boomboomsatellites! Thanks for reporting that issue. Just a view questions:
* Which example *.ino do you have used?
* Which Hardware do you use
* Does the LED colors are changing?
* Do you have connected your Boost Hub via the official LEGO APP?

I will check my hub firmware version and give feedback if i am back at home

corneliusmunz, 2019-11-16:
@boomboomsatellites I have checked my firmware and Hardware version and i have a newer one. Lego has changed port assignments and maybe thats the problem in turning your motors. Just connect with the Boost App to your hub and then the firmware update will start. After the firmware ist flashed to the current version the examples in the library should work

Firmware version major:2 minor:0 bugfix:0 build:23
Hardware version major:0 minor:4 bugfix:0 build:0

You can check your firmware version in the serial monitor if you add the following function call after the hub is connected:
`myBoostHub.activateHubUpdates();`

To get log messages in the serial monitor you have to add a preprocessor statement in your *.ino file just at the top of your code
`#define LOGGING_ENABLED`

boomboomsatellites, 2019-11-18:
Sorry for being late.

- Which example *.ino do you have used?
 BoostHub,PowerdUpRemoteBoost.ino

- Which Hardware do you use
 Lego Boost&M5stickC
https://www.lego.com/ja-jp/service/device-guide/boost
https://m5stack.com/products/stick-c

- Does the LED colors are changing?
 The LED did not change.It was a misunderstanding

- Do you have connected your Boost Hub via the official LEGO APP?
 Yes. Connected to the latest version of LEGO Boost APP (iOS) and updated the firmware.

I tried to get the log, but the version is the same ...

data: 9 0 1 3 6 17 0 0 20 
Firmware version major:2 minor:0 bugfix:0 build:23
data: 9 0 1 4 6 0 0 0 4 
Hardware version major:0 minor:4 bugfix:0 build:0

Is there anything else that can be examined here?

corneliusmunz, 2019-11-19:
@boomboomsatellites Strange... i have checked it again with the current release and current firmware and the BoostHub.ino example file and it works fine. I think the BLE connection works on your side because you receive Log messages with the Firmware and Hardware version. So i am a little bit confused and have no idea what is going wrong. 

Can you do the following procedure:
* power up your arduino hw
* flash/upload the BoostHub.ino file
* power up your boost hub
* press the reset button on your arduino

boomboomsatellites, 2019-11-21:
Hello @corneliusmunz

The Arduino currently in use does not work.

I'll try using another Arduino.

It was over trouble.

It will end.

closed 2019-11-21 by boomboomsatellites