corneliusmunz/legoino #10: Impossibile to connect more than 3 hubs

GianCann, 2020-02-18:
I have 10 PUP Smart Hubs, but i don't have success to connect more than 3 hubs simultaneously.
When i try to connect the fourth hub, the program stop execution...

GianCann, 2020-02-19:
I've a lot of hubs to test the library ;)
[image]

corneliusmunz, 2020-02-19:
@GianCann Wow... thats a cool Setup! Beside the issue i am really interested what type of usecase you have. But now to the issue:

I think it is related to the esp32 board/hw configuration where you can define the maximum number of allowed BLE connections. This is by default set to :
https://github.com/espressif/arduino-esp32/blob/13e02063d63146096952fc6a6ef0bc3ced5b1203/tools/sdk/sdkconfig#L186

I found some issue reports where the same issue is tackled and maybe you are able to change that setting and compile a new hw config for arduino:
https://github.com/espressif/arduino-esp32/issues/2730

Unfortunately the BLE Spec of the ESP32 states that the max number of connections is limited to 7 (page 7 of the document):
https://www.espressif.com/sites/default/files/documentation/esp32_bluetooth_architecture_en.pdf

But other reports tell that are e.g. 9 connections possible (or maybe even more):
https://github.com/espressif/esp-idf/issues/2717

Maybe you can try to change the hw config to the following settings and can try it again. Unfortunately i have not so many devices as you and can not test it by myself ;-)

`
CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN=8
CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF=8
`

corneliusmunz, 2020-02-19:
@GianCann  maybe you can set the debug level of your esp device in the arduino environment to VERBOSE. If you have changed it i am interested in the log output when it crashes...
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md#switching-on-debugging

corneliusmunz, 2020-02-19:
@GianCann Sorry for spamming you with comments... If it is not possible to go above max 8 connections maybe we can think of different esp's which have a wifi or socket connection and one of the esp's is the master and the other are the slaves

GianCann, 2020-02-19:
@corneliusmunz  it's not a problem ;)
Anyway, 8 device it's a good chance to manage a train layout ;)
Can you tell me where I can chance the CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN parameter?

GianCann, 2020-02-19:
Ok, I need to recompile the entire SDK for ESP32-Arduino...
I will tryed to do ;)

corneliusmunz, 2020-02-24:
@GianCann I tried to recompile the esp32 sdk and changed the BLE configuration but unfortunately i cannot connect more than 3 controllers. I am not shure if i have done all the recompilation in the right way and i am not shure if the arduino IDE uses really the recompiled *.a libraries. I will dig a littlebit deeper into that issue and try to solve it

GianCann, 2020-02-26:
Hi @corneliusmunz , thank to my friends @ivanporro, I also managed to recompile the esp32-arduino library, modifying these parameters

CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN=9
CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN__EFF = 9
CONFIG_BT_ACL_CONNECTIONS = 7

Now, what other steps did you follow?

ivanporro, 2020-02-26:
Hi @corneliusmunz  I did a step further.

I've managed to include arduino libs plus Legoino as libraries inside a project compatible with the ESP SDK structure. 
I've added the library sources as for other libraries then modified the includes in CMakeLists.txt to make sure al the cpp files from your lib are included.
I'am about to have a working compiled .elf for an ESP32, however it dies close to the end. The "issue" is related to the ESP SDK, which is in C. By moving the main to a .cpp format and doing some hacks here & there I'm almost done but with some errors in your library. E.g.:

../components/arduino/libraries/Legoino/src/Lpf2Hub.cpp:140:52: error: narrowing conversion of '(size + 2)' from 'int' to 'byte {aka unsigned char}' inside { } [-Werror=narrowing]

../components/arduino/libraries/Legoino/src/Lpf2Hub.cpp:288:9: error: 'i' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         for (int i; i < charArrayLength; i++)

Any hint?
Thanks!

ivanporro, 2020-02-26:
Well, just for the record we managed to compile it but suppressing some warning being treated as error in::

...../esp-idf/tools/cmake/idf_functions.cmake

list(APPEND compile_options "-Wall"
                                "-Werror=all"
                                "-Wno-error=unused-value"
                                "-Wno-error=maybe-uninitialized"
                                "-Wno-error=narrowing"

Now, the ESP is panicking and not loading the sketch which may or may be not related to those warning.

GianCann, 2020-02-27:
Hi @corneliusmunz 

> I tried to recompile the esp32 sdk and changed the BLE configuration but unfortunately i cannot connect more than 3 controllers.

Can you tell me your steps? I want to verify with my owns.

GianCann, 2020-08-05:
After some time...
https://www.youtube.com/watch?v=vOMIabLJhak

Here the solution:
https://github.com/h2zero/NimBLE-Arduino/issues/99#issuecomment-668690196

corneliusmunz, 2020-09-05:
@GianCann With the release version 0.7.0 and the switch to the NimBLE-Library (version 1.0.1) it is possible to connect to up to 9 devices. Thanks again for pointing me in the direction of the new NimBLE-Library and for testing it with up to 9 hubs. Looking forward to your cool projects :-)

corneliusmunz, 2020-09-15:
I have added a short documentation what is needed to connect to more than 3 hubs: https://github.com/corneliusmunz/legoino#connection-to-more-than-3-hubs

closed 2020-09-15 by corneliusmunz