corneliusmunz/legoino #11: Status --> is connected for train hub

Hoelle1010, 2020-04-16:
Hello Cornelius,

first thanks lot lot for your fantastic work on the legion library. It works super for my train hubs.

I have one question. I want to implement a LED on the ESP32 to show if and which train is connected. Unfortunately the command "isConnected" shows me just the lastet status. So if I turn off the train hub the "isConnected" still is "1 or true". Is there any way to check if the connection ist available anymore.

Thank a lot and regards
Hölle

corneliusmunz, 2020-04-19:
Hi @Hoelle1010! Thanks for your issue! Thats a good request. I think this could be handled with the disconnect event of the BLE library which is be used by the legoino lib: https://github.com/nkolban/ESP32_BLE_Arduino/blob/adc2aee2f0d01eb6b30dd5ad3589f2cc89934beb/src/BLEClient.cpp#L190 

I will check it and evaluate if i can add a callback function if a disconnect appears. You are also free to do a pull request which could be reviewed by me. 

So i will have a look in the next days and try to solve that issue

corneliusmunz, 2020-04-22:
Hi @Hoelle1010! I have evaluated that issue and have done a quick fix. Now the disconnect callback of the BLEDevice is used to update the ```isConnected()``` status. You can try it out with the newest [0.6.2](https://github.com/corneliusmunz/legoino/releases/tag/0.6.2) release. This should be available in the arduino library manager overnight. Hope this will work for you. You can check the behaviour with the ```TrainHub.ino``` example sketch by connecting and after the connection do a long press of the hub button to force the shutdown/disconnect of the hub. Maybe you can give me a short feedback if it works.

corneliusmunz, 2020-04-27:
Hi @Hoelle1010 any feedback on that issue. Is it resolved with the latest release?

Hoelle1010, 2020-04-27:
Hello Cornelius,

This was very fast and it is working perfect for me.
Many, many thanks.

Regard Hölle

closed 2020-04-27 by Hoelle1010

corneliusmunz, 2020-04-27:
@Hoelle1010 perfekt! Thanks for your fast feedback and the closure of the issue. If you like you can star the repo :-)

Hoelle1010, 2020-04-27:
Yes, of course. Unfortunately I am new at github, so I do not know all functions, yet.