corneliusmunz/legoino #15: Library compatibility problem

colas-sebastien, 2020-06-02:
Hi,

The Legoino library is not compatible with the M5Stack / M5StickC Library due to re-declaration of colors.
I propose to rename the color in the library like that (file `Lpf2Hub.h`) :
```
typedef enum Color
{
  LEGO_BLACK = 0,
  LEGO_PINK = 1,
  LEGO_PURPLE = 2,
  LEGO_BLUE = 3,
  LEGO_LIGHTBLUE = 4,
  LEGO_CYAN = 5,
  LEGO_GREEN = 6,
  LEGO_YELLOW = 7,
  LEGO_ORANGE = 8,
  LEGO_RED = 9,
  LEGO_WHITE = 10,
  LEGO_NONE = 255
};
```

corneliusmunz, 2020-06-02:
Hi @colas-sebastien ! Thanks for the hint. Can you copy me the compiler error and show me where the  declaration in the M5Stack/Stick Library is?

colas-sebastien, 2020-06-02:
Hi @corneliusmunz ,
It seems the problem only occurred with the M5StickC library.
Here is my use case:
```
#include <M5StickC.h>
#include <PoweredUpHub.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

The output of the compiler:
```
In file included from /home/programmez/Arduino/libraries/M5StickC/src/utility/In_eSPI_Setup.h:17:0,
                 from /home/programmez/Arduino/libraries/M5StickC/src/utility/In_eSPI.h:24,
                 from /home/programmez/Arduino/libraries/M5StickC/src/M5Display.h:7,
                 from /home/programmez/Arduino/libraries/M5StickC/src/M5StickC.h:70,
                 from /tmp/arduino_modified_sketch_117376/sketch_jun02a.ino:1:
/home/programmez/Arduino/libraries/M5StickC/src/utility/ST7735_Defines.h:94:29: error: expected identifier before numeric constant
 #define BLACK               0x0000      /*   0,   0,   0 */
                             ^
/home/programmez/Arduino/libraries/Legoino/src/Lpf2Hub.h:77:3: note: in expansion of macro 'BLACK'
   BLACK = 0,
   ^
/home/programmez/Arduino/libraries/M5StickC/src/utility/ST7735_Defines.h:94:29: error: expected '}' before numeric constant
 #define BLACK               0x0000      /*   0,   0,   0 */
                             ^
/home/programmez/Arduino/libraries/Legoino/src/Lpf2Hub.h:77:3: note: in expansion of macro 'BLACK'
   BLACK = 0,
   ^
/home/programmez/Arduino/libraries/M5StickC/src/utility/ST7735_Defines.h:94:29: error: expected unqualified-id before numeric constant
 #define BLACK               0x0000      /*   0,   0,   0 */
                             ^
/home/programmez/Arduino/libraries/Legoino/src/Lpf2Hub.h:77:3: note: in expansion of macro 'BLACK'
   BLACK = 0,
   ^
In file included from /home/programmez/Arduino/libraries/Legoino/src/PoweredUpHub.h:14:0,
                 from /tmp/arduino_modified_sketch_117376/sketch_jun02a.ino:2:
/home/programmez/Arduino/libraries/Legoino/src/Lpf2Hub.h:89:1: error: expected declaration before '}' token
 };
 ^
exit status 1
Error compiling for board M5Stick-C.
```

corneliusmunz, 2020-06-03:
Hi @colas-sebastien ! Thanks for sharing the code and warning 👍 . Now it is clear. The MStickC Lib had defined some preprocessor macros: 
`#define BLACK               0x0000      /*   0,   0,   0 */`
https://github.com/m5stack/M5StickC/blob/1befa455553c965aeb7c91e8515a9f192310f7b4/src/utility/ST7735_Defines.h#L94

This macros will be replacing all occurences of color names with the values on the right hand side.  Now the question is if legoino or M5stickc lib should be changed 😄  I will think about how to overcome the problem.  Maybe i use your proposel in renaming th colors.

corneliusmunz, 2020-06-03:
@colas-sebastien  I have also opened an issue on the m5stick lib because there are namespaced color defines available and in my point of view the color defines which causes the problem could be replaced on the m5 side. 
Opened Issue: https://github.com/m5stack/M5StickC/issues/113

colas-sebastien, 2020-06-03:
OK thanks for that,
let me know when the problem is solved.
For the moment I use my solution by renaming color in leoino enum.

colas-sebastien, 2020-07-26:
Hi Cornelius,

My code that use you library is available on my github repo:

https://github.com/colas-sebastien/M5StickC-joystick-lego-powerup-batmobile

Regards,
Sébastien

corneliusmunz, 2020-07-26:
@colas-sebastien Very cool! Do you have a short video?

closed 2020-08-31 by corneliusmunz