corneliusmunz/legoino #16: Support for Control+

jabrena, 2020-06-27:
Hi, 

Is the library compatible with LEGO control+?

corneliusmunz, 2020-07-09:
Hi @jabrena ! I‘m working on the support for Control+ but it will take some time. But in the meanwhile you can use the work of the following Pull-Request. I will not merge it because I will structure it a little bit different but it works: https://github.com/corneliusmunz/legoino/pull/6

corneliusmunz, 2020-09-05:
@jabrena With the new release version 0.7.0 the support of the Control+ Hubs are added: https://github.com/corneliusmunz/legoino/releases/tag/0.7.0

jabrena, 2020-09-06:
Many thanks,

I received the board, but I was on holidays and I could deploy any code in the board. Next week, I will give you feedback.

Cheers

closed 2020-09-15 by corneliusmunz