corneliusmunz/legoino #17: Support for "StartSpeed" sub command 0x07/0x08

ThorstenBenter, 2020-07-31:
Der Cornelius,
first of all, thank you very much for all your efforts and the really nice software your have created. Because I became aware of legoino, I got myself an ESP32 dev board - installed all the things required and it just was going like a charm: Can control the PoweredUp hub (2I/O) using your software.
I also got myself the LEGO crocodile, and a Pup L Motor. I am programming my own pieces of software using VB6 (no joke) with the nsoftware BLE stack (I got that for free from them, when I told them what I was planning to do ... nice move!)
So far I can control 10+ hubs (Technic hubs (4I/O) and Hubs (2I/O). All on a very basci level, as I am no programmer at all - I'm a chemist :-)
My question is, whether you are planning on implementing the StartSpeed sub command (0x07/0x08) of the output command 0x81? I believe you have done that for the Boost controller already. The rather new Pup L motor (in contrast to the train motor) has a built-in rotation sensor. I used that feature to run my Crocodile - an it goes really smoothly. Nice to see it moving slowly but at constant speed. I simply used the LWP StartSpeed (Speed, MaxPower, UseProfile) command with MaxPower 100(%) and UserProfile = 0 (no profile).
My current project is to equip the piece of track on the wall with some sensors, and then sue the ESP32 along with your software to let it go back and forth; here is the idea: https://www.eurobricks.com/forum/index.php?/forums/topic/179240-mod-10277-and-on-and-on-another-crocodile-mod-%E2%80%A6/
All the best
Thorsten

corneliusmunz, 2020-07-31:
@ThorstenBenter Hi Thorsten! Cool project and VB6 is totally fine :-) I will have a look on the specific command for the Pup L Motor and will try it out. I will give you feedback if it works. If it works i will update the library and release a new version

corneliusmunz, 2020-10-06:
@ThorstenBenter Hi Thorsten! I am working on the release of version 1.0.0 and there it will be possible to use the Tacho Motor commands also with the Two Port Hub (PoweredUp Train Hub).  I think during this week i will be able to make it ready for a release. There will be some breaking changes but i will write a migration guide. Some functions will be renamed and the global variables will be removed and will be replaced by callback functions for all the sensor values

ThorstenBenter, 2020-10-07:
@corneliusmunz Hi Cornelius!
These are very, very nice news! I will be very happy to migrate - as it is so much fun to work with your code. I made a simple workaround by "ramping" the power values up and down. But having PID speed control will be a game changer for all LEGO models, particularly trains when operated with the L-type motor. As the PF-L and the PUp-L motor have the same physical dimensions, one can simply change from PF to PUp and then enjoy very, very smooth operation of e.g. trains at low speed.
Thank you for your all your efforts in providing the community with such nice software!
All the best,
Thorsten

corneliusmunz, 2020-10-19:
Fixed with Release 1.0.0 https://github.com/corneliusmunz/legoino/releases/tag/1.0.0

closed 2020-10-19 by corneliusmunz