corneliusmunz/legoino #18: Is it possible to get tacho motor data?

jabrena, 2020-08-15:
I would like to know if it is possible to read tacho in the same way that this library:
https://github.com/nathankellenicki/node-poweredup/issues/32

Regards 

Juan Antonio

corneliusmunz, 2020-08-18:
Hi @jabrena I have one example (BoostHubRotationSensor.ino) for the Boost Tacho Motor but it should work in a similar way for the Control+ Hub with a connected Tacho Motor: https://github.com/corneliusmunz/legoino/blob/edc40fa058e640bf455411be0708832a0e13b093/examples/BoostHubRotationSensor/BoostHubRotationSensor.ino#L47

jabrena, 2020-08-18:
This week, I will receive the board:
https://store.arduino.cc/arduino-nano-33-ble

[image]

I would like to test the library with the board and test the support with Control Hub:

[image]

Cheers

Juan Antonio

corneliusmunz, 2020-08-18:
Hi @jabrena I think it will not work out of the box because the library supports up to now only boards with the Esp32 chip 🙁 I have to check if the official Arduino BLE lib will also work. Looking forward to your experiences

jabrena, 2020-08-19:
SeeKool Tablero de Desarrollo ESP-32 ESP-32S WiFi + Bluetooth 2-en-1 Dual Core Módulo de Antena de 2.4 GHz con Antena RF Amp Filtro Ap STA Arduino IDE, Ultra-Low Power (1 pcs) (1 Pc) https://www.amazon.es/dp/B07DPP3BGZ/ref=cm_sw_r_cp_api_i_BbvpFbJK7BRP1

corneliusmunz, 2020-08-19:
@jabrena Yes, this board is compatible

jabrena, 2020-08-19:
I have just ordered it.

In the other hand, I would like to understand the response from Technic Hub:

```
17:55:27.375 -> Service 1801
17:55:27.375 -> 	Characteristic 2a05, properties 0x20
17:55:27.375 -> 		Descriptor 2902, value 0x0018
17:55:27.375 -> Service 1800
17:55:27.375 -> 	Characteristic 2a00, properties 0x4E, value 0x546563686E696320487562
17:55:27.410 -> 	Characteristic 2a01, properties 0x4E, value 0x0018
17:55:27.444 -> 	Characteristic 2a04, properties 0x2, value 0x0018
17:55:27.477 -> Service 00001623-1212-efde-1623-785feabcd123
17:55:27.477 -> 	Characteristic 00001624-1212-efde-1623-785feabcd123, properties 0x1E, value 0x0F000403012F0000100000001000000000000000
17:55:27.511 -> 		Descriptor 2902, value 0x00
```

- What is the purpose of the 3 different services?
- If I want to change the led color in Control hub, in what Service Should I have to send?
- What is the concept about Characteristics? In the context of Control hub, what is purpose of the characteristics?

I try to understand and learn

Many thanks in advance.

Cheers

Juan Antonio

corneliusmunz, 2020-08-19:
Hi @jabrena ! Services and Characteristics are BLE terms of the so called GATT Bluetooth Low energy protocol. You can find a good overview on the following site: https://learn.adafruit.com/introduction-to-bluetooth-low-energy/gatt The last service and characteristic is the Lego powered up characteristic for nearly all hubs (except the wedo hub)

closed 2020-09-15 by corneliusmunz

tintamarre, 2021-01-19:
For reference, you can use this project : https://github.com/hoharald/leguno-remote