corneliusmunz/legoino #19: Is it possible to auto-detect connected devices?

lucadentella, 2020-08-29:
Hi! 
Is it possible to detect what is connected to a specific port? 
I'm playing with Technic Hub and the official Lego app is able to perform a connection-check when selecting the model to control.
I thought I could use the **getDeviceTypeForPortNumber()** method but it always return "0" even if a motor is connected to the specified port and I can control it with **setMotorSpeed** commands.

Thanks

GianCann, 2020-08-29:
Hi @lucadentella ,
which type of motor you have connected to the hub?

GianCann, 2020-08-29:
Here you can find more "Device Id" related to LPF2 sensor/motor that not are listed in getDeviceTypeForPortNumber()

https://github.com/GianCann/technical-info/blob/master/assigned-numbers.md

GianCann, 2020-08-29:
Take in mind that the LWP Protocol implement a event notification when you attach/detach a device from the Hub:
https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#hub-attached-i-o

lucadentella, 2020-08-31:
Thanks! I connected both Control+ XL Motor and Control+ Large Motor (the ones included in Technic Control+ Sets) but I get "0" for both of them.

I enabled the logging feature of the library and I discovered something "strange". 
The code of `byte Lpf2Hub::getDeviceTypeForPortNumber(byte portNumber)` loops on this variable: **numberOfConnectedDevices** but when the library prints out its value (`LOGLINE(numberOfConnectedDevices, DEC);`) it's always zero...

GianCann, 2020-08-31:
@lucadentella when a device is connected to the hub, the hub send a Port Message (0x04) to inform that a device is attached or detached.
When is an 'attached' message, the message contains some info about the device: the Device Id Type, the port where is attached and the hardware firmware version.
See the 'notifyCallback' function, that call the parsePortMessage.

I think that the problem is in parsePortMessage function, that lack to manage numberOfConnectedDevices  and Device Type ID.

corneliusmunz, 2020-08-31:
@GianCann you are right. The attach message is evaluated in the following function:
https://github.com/corneliusmunz/legoino/blob/edc40fa058e640bf455411be0708832a0e13b093/src/Lpf2Hub.cpp#L423
But unfortunately the function only logs a message and does not call the following method which "stores" the attached device in a local struct array with the device information. I will check that and correct the lib
https://github.com/corneliusmunz/legoino/blob/edc40fa058e640bf455411be0708832a0e13b093/src/Lpf2Hub.cpp#L256

If a disconnect event occurs the parsePortMessage method should also call deactivatePortDevice to disconnect the device and remove the device from the struct array:
https://github.com/corneliusmunz/legoino/blob/edc40fa058e640bf455411be0708832a0e13b093/src/Lpf2Hub.cpp#L270

GianCann, 2020-08-31:
@corneliusmunz  Unfortunately, in these days, I have a lot of things to do otherwise I would helped you to improve the library
Off topic: @lucadentella sono quasi certo di averti scritto una email o un messaggio tramite il tuo sito, nelle passate settimane riguardo ESP32 e il limite delle connessioni BLE...

lucadentella, 2020-08-31:
@corneliusmunz I added some printouts to the `parsePortMessage` method and, as soon as the hub is connected, I receive the following messages which seem correct (I connected two Large Motors - code 2E - to port A and B)

```
Port 0 status: CONNECTED to device type 2E
Port 1 status: CONNECTED to device type 2E
Port 32 status: CONNECTED to device type 17
Port 3B status: CONNECTED to device type 15
Port 3C status: CONNECTED to device type 14
Port 3D status: CONNECTED to device type 3C
Port 60 status: CONNECTED to device type 3C
Port 61 status: CONNECTED to device type 39
Port 62 status: CONNECTED to device type 3A
Port 63 status: CONNECTED to device type 3B
Port 64 status: CONNECTED to device type 36
```

GianCann, 2020-08-31:
Yes, it's all correct.
The other device are the internal sensor (battery, current, gyro) and LED Light ;)

lucadentella, 2020-08-31:
only one dubt... Powered Up hub IMU temperature (type 3C) seems connected to two ports (3D and 60).
BTW I wasn't able to find your message regarding esp32... :/

lucadentella, 2020-08-31:
only one dubt... Powered Up hub IMU temperature (type 3C) seems connected to two ports (3D and 60).
BTW I wasn't able to find your message regarding esp32...

GianCann, 2020-08-31:
> only one dubt... Powered Up hub IMU temperature (type 3C) seems connected to two ports (3D and 60).

Which hub you hare using?

> BTW I wasn't able to find your message regarding esp32... :/

Ok, no problem (in the meantime, we have resolved the issue: https://github.com/corneliusmunz/legoino/issues/10 )

GianCann, 2020-08-31:
Note: If you use the Technic Hub (aka Control+ HUB) it have two temperature sensor. One internal on the STM32, and one external.

corneliusmunz, 2020-09-05:
@lucadentella the `getDeviceTypeForPortNumber` function should work on a feature branch. Would be great if you could check it with that feature branch and possible give me feedback. I have checked it on my side and you can now fetch the devices on a specific port even if several hubs are connected. If you have checked it i will merge the branch into the main branch and generate a new release. The feature branch where you can check it is the following: https://github.com/corneliusmunz/legoino/tree/feature/check_NimBle_Library

corneliusmunz, 2020-09-05:
@lucadentella The issue should be fixed with the new 0.7.0 release: https://github.com/corneliusmunz/legoino/releases/tag/0.7.0

lucadentella, 2020-09-07:
@corneliusmunz confirm, I've tested now! Thanks for your effort, I'm closing this issue. 

BTW a suggestion: the new version requires NimBLE library, you could add it as a **dependency** in your manifest so that when a user install Legoino from the Library Manager, it automatically suggests also to install NimBLE.

closed 2020-09-07 by lucadentella

corneliusmunz, 2020-09-07:
@lucadentella Super! Thanks for testing and very good hint with the link to the dependency. I was not aware of that feature.

corneliusmunz, 2020-09-07:
@lucadentella  I have released a new version 0.7.1 where i have added the dependent library in the properties file. Now the Arduino Library manager will detect the dependent library automatically