corneliusmunz/legoino #22: Lego Duplo PoweredUp trains

cougarek, 2020-09-28:
Are Lego Duplo trains 10874 and 10875 supported?

corneliusmunz, 2020-09-28:
Hi @cougarek! Yes, in principal this should be possible by using the PoweredUpHub class (Train hub) to control the connected motor (use port A (0x00) as port) For the Speedometer, Speaker and Color Sensor i think the library need some adaptions. Unfortunately i have no Duplo train set available and therefore it is difficult for me to test. I will have a look on the specification about Speaker and Color Sensor and maybe you can do some tests for me.

cougarek, 2020-09-29:
Hi, sure, I can do some tests. I have ESP32 module coming in two days.

cougarek, 2020-10-01:
I just flashed ESP32 with TrainHub.ino example. The hub is connecting, the motor is working, but LED is not changing color. I also tried TrainColor.ino example but color sensor is returning 0 all the time. Are there any tests I can do for you?

corneliusmunz, 2020-10-03:
Hi @cougarek 

regarding the Hub LED you can try to use the following commands. The Port number ob the Hub LED for the Duplo Train hub is 0x12 (18 decimal) and for the other Hubs like Boost Hub or Train Hub it is 0x32 (50 decimal). So i have to add an extra class for the DuploTrainHub. But you should be able to set the color with the following lines manually:

```
byte setColorMode[8] = {0x41, 0x12, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00};
myTrainHub.WriteValue(setColorMode, 8);
byte setColor[6] = {0x81, 0x12, 0x11, 0x51, 0x00, color};
myTrainHub.WriteValue(setColor, 6);
```

Alternatively to use the `setLedColor` command, you can change the function locally in your `/libraries/legoino/src/Lpf2Hub.cpp file` and replace the code with the following one (just as a workaround for you till i will made the changes for the Duplo hub)

```
/**
 * @brief Set the color of the HUB LED with predefined colors
 * @param [in] color one of the available hub colors
 */
void Lpf2Hub::setLedColor(Color color)
{
    byte setColorMode[8] = {0x41, 0x12, 0x00, 0x01, 0x00, 0x00, 0x00, 0x00};
    WriteValue(setColorMode, 8);
    byte setColor[6] = {0x81, 0x12, 0x11, 0x51, 0x00, color};
    WriteValue(setColor, 6);

    0x12 Duplo Hub 
    
}

```
All the sensors which are attached to a port has to be activated for sending updated values. For the Duplo Train Color sensor it seems that it has a different setup (port and device type) than the one in the trainColor.ino example. Can you please adapt the example with the following line and test it again?

`myHub.activatePortDevice(0x12, 0x2B);`

This line will activate the color sensor on Port 0x12 for the Device Type 0x2B which is the Duplo Train Color sensor. But maybe i have to change something in the reading of the requested color sensor value. Would be great if you can test the changes because i have no Duplo Train Hub :-)

cougarek, 2020-10-04:
@corneliusmunz Unfortunately no luck yet :) 

Setting color doesn't work - LED stays blue all the time. I tried with manual command and by changing code in `Lpf2Hub.cpp` file.

Regarding reading color sensor value I activated the sensor and still getting zeros. Btw for `TrainColor.ino` example I had to move `myHub.init();` to main loop to make it connect to Hub (same init code as `TrainHub.ino` example).

cougarek, 2020-10-04:
Update: There is a way to turn on and off LED light:
set mode `{0x41, 0x01, 0x01, 0x01, 0x00, 0x00, 0x00, 0x01}`
on `{0x81, 0x11, 0x11, 0x51, 0x00, 0x0a}`
off `{0x81, 0x11, 0x11, 0x51, 0x00, 0x00}`
After issuing above commands LED changes color to default white. `setLedColor` command still doesn't work.

We can also play predefined sounds with same set mode: (probably there is more of them)
horn `{0x81, 0x01, 0x11, 0x51, 0x01, 0x09}`
fill water `{0x81, 0x01, 0x11, 0x51, 0x01, 0x07}`

corneliusmunz, 2020-10-06:
Ok... strange. I think i have to buy a set to test it at home 😉

corneliusmunz, 2020-10-06:
@cougarek Order for the 10874 set is done... i will get it on thursday and could hopefully provide you a solution where the Legoino lib will also support the Duplo sets with all commands (motor, light, sound, color sensor). Looking forward to the delivery 😀

corneliusmunz, 2020-10-08:
Test setup is ready 😀
[image]

corneliusmunz, 2020-10-08:
@cougarek I have implemented and checked all the available functions. In the next release (hopefully during the next weekend) the support of the Duplo Train hub will be included. You can test it before the release in checking out the following feature branch of the library: https://github.com/corneliusmunz/legoino/tree/feature/add_hub_emulation Here you can find an DuploTrainBase.ino example sketch where you can see how the sensor reading will work. Just put your Train on the Rails and push it a littlebit forward or backward. If you lift it from the track or hold it, it will stop. The color sensor readings should be directly visible in the Hub LED and dependent on the color a sound will be played. 

* Speedometer could be read out
* Color sensor could be read out
* Motor Commands (Basic Motor)
* LED Color (color, rgb, hsv)
* Sound (basic sounds and some system tones)

So just have a look on the feature branch and i would be very happy if you could test it and could give me feedback. 

Btw. The LED Port was not `0x12`it was `0x11` :smile:

cougarek, 2020-10-09:
Thank you for your commitment :) 
I don't have a lot of time to test at the moment but it looks good, except for the reading of colors. When I put train gently on white piece of paper it gives me some random readings, example:
```
Color: yellow
Color: blue
Color: 
Color: black
Color: 
Color: black
Color: black
Color: cyan
Color: blue
Color: yellow
Color: white
```
But I noticed last reading is almost always right, but more tests need to be done.

corneliusmunz, 2020-10-19:
Fixed with Release 1.0.0 https://github.com/corneliusmunz/legoino/releases/tag/1.0.0

closed 2020-10-19 by corneliusmunz