corneliusmunz/legoino #26: Hub emulation dependency on NimBLE???

AlbanT, 2020-10-20:
I installed the new 1.0.0 version of Legoino and while testing the HubEmulation example I got a compilation error on NimBLEDevice.h.
Maybe I missed a warning about dependencies but after I installed NimBLE-Arduino library it worked fine.
If it isn't in the documentation (I'm dyslexic so maybe I just missed it) I think it should be added :)

corneliusmunz, 2020-10-20:
Hi @AlbanT ! Thanks for raising that issue. I have described it in the following section: https://github.com/corneliusmunz/legoino#remarks and in the Quickstart tutorial. Additionally i have marked it for the Library Manager as a dependent library and it should appear if you install Legoino that the dependent Library is needed.

But nevertheless. The hint about the dependency should appear at the beginning of the documentation and not at the end. So i will change this

corneliusmunz, 2020-10-21:
I have improved the documentation and moved the information about the dependent library at the top of the README.md file. This is fixed with the following release: https://github.com/corneliusmunz/legoino/releases/tag/1.0.1

closed 2020-10-21 by corneliusmunz