corneliusmunz/legoino #27: New HubEmulation only works on my iPhone

AlbanT, 2020-10-20:
Hi,

I managed to upload the hub emulation example to my ESP32 (Lolin32 V1.0.0) and the Lego PoweredUp app (V3.5.0) on my iPhone connected without any issues.
When my son tried to connect with the Lego PoweredUp app (V3.5.0) on his Android phone or the Galaxy Tab the connecting fails.
Is this an isolated issue or can anyone reproduce this?
I havn't tried an earlier version of the Lego PU app yet.

corneliusmunz, 2020-10-20:
@AlbanT Good point! I have never tested it on an Android phone. I will do this with my wifes phone and will have a look on the debug messages. It is definitely a BETA feature but also definitely a feature which is based on your initial Idea 👍 So i will have a look and give you feedback. Maybe you can set the Arduino Log Level to DEBUG (Tools  menu, Core debug level) and open the serial monitor while you try to connect. It would be very helpful for me if you can paste then the output here in the issue as a comment.

AlbanT, 2020-10-20:
I managed to turn on the log level to DEBUG. Was not that simple as the **WEMOS LOLIN32** doesn't have that option but as it turns out I can compile and upload as an **ESP32 Dev Module** which has the debug option.

Unfortunately the debug log doesn't tell us anything as the PU App never gets to even detecting the ESP let alone connect :(
So the log stays idle at:

`[D][Lpf2HubEmulation.cpp:449] start(): characteristic defined! Now you can read it in your phone!`

corneliusmunz, 2020-10-20:
@AlbanT I have tried it now with an Android phone an have experienced the same issue as you. But interestingly i tried to connect to a "real" Train hub and even that does not work. Can you test it if you can connect to a real hub?

AlbanT, 2020-10-20:
Unfortunately not. All I have is a Boost hub and a Technic hub :(
A few days ago I did experience connection issues with my iPhone to the Technic hub and in that particular case my Galaxy tab was able to connect. So maybe there is some sort of issue in the Lego app?

corneliusmunz, 2020-10-21:
I have evaluated the issue an the so called advertisment data is incomplete. The iOS operating system is maybe a littlebit more tolerant than Android. I will try to fix it as soon as possible

corneliusmunz, 2020-10-21:
Hi @AlbanT ! I have changed some settings in the advertisement data and now i could connect with an android phone to the emulated hub. It would be very helpful if you can cross check if it works on your side. The fix is available in the following release: https://github.com/corneliusmunz/legoino/releases/tag/1.0.1

AlbanT, 2020-10-22:
Hi @corneliusmunz I have installed V1.0.1 of Legoino (using the library manager in Arduino IDE) and recompiled the sketch.
The issue persists on my Android tablet and my sons phone but I do get a bit further. The Lego PU app does show the ESP32Hub (I renamed it in the sketch) but it still won't connect. Yesterday with V1.0.0 the app only showed an empty block without the in the sketch specified name.
Did you try it with the same version of the app as I did? I'm currently at 3.5.0.

I made a short clip to illustrate: [youtube](https://youtu.be/nn4pjtCqByM)

corneliusmunz, 2020-10-22:
Hi @AlbanT ! Hm... can you try the following things:
* can you "kill" the app and restart it? 
* Have you installed the latest version 1.0.2 of the NimBLE-Arduino library?
* Can you write me which Android versions you use? 
* Can you use a shorter Hub Name?
* Can you enable the NimBLE logs with the following changes?

1) open the following file (OSX: `~/Documents/Arduino/libraries/NimBLE-Arduino/src/nimconfig.h`, Windows: `.../My Documents/Arduino/libraries/NimBLE-Arduino/src/nimconfig.h`) and open the NimBLE-Arduino folder? 
2) edit `nimconfig.h`and uncomment line 74 (`#define CONFIG_BT_NIMBLE_DEBUG`) 
3) Restart Arduino to force the rebuild of the solution
4) Open serial Monitor after the code is uploaded to your device
5) Copy the log and paste it here in the comments

This would be very helpful. Thank you so much for your patience :smile:

AlbanT, 2020-10-22:
Hi @corneliusmunz 

- killed the app every time as that was the only way out again :)
- I have v1.0.2 of the NimBLE library according to the Arduino library manager
- Both the tablet and my sons phone uses Android 7.0 
- Renamed the hub to simply **Hub**
- Enabled the NimBLE logging this is the resulting log:
```
17:24:24.833 -> ets Jun  8 2016 00:22:57
17:24:24.833 -> 
17:24:24.833 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:24:25.245 -> ets Jun  8 2016 00:22:57
17:24:25.245 -> 
17:24:25.245 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:24:25.245 -> configsip: 0, SPIWP:0xee
17:24:25.245 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:24:25.245 -> mode:DIO, clock div:1
17:24:25.245 -> load:0x3fff0018,len:4
17:24:25.245 -> load:0x3fff001c,len:1216
17:24:25.245 -> ho 0 tail 12 room 4
17:24:25.245 -> load:0x40078000,len:9720
17:24:25.245 -> ho 0 tail 12 room 4
17:24:25.245 -> load:0x40080400,len:6352
17:24:25.245 -> entry 0x400806b8
17:24:25.486 -> [D][Lpf2HubEmulation.cpp:380] start(): Starting BLE
17:24:25.729 -> I NimBLEDevice: "BLE Host Task Started"
17:24:25.729 -> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0003 len=0
17:24:25.729 -> 0x03 0x0c 0x00 
17:24:25.729 -> Command complete: cmd_pkts=5 ogf=0x3 ocf=0x3 status=0 
17:24:25.729 -> ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0001 len=0
17:24:25.729 -> 0x01 0x10 0x00 
17:24:25.729 -> Command complete: cmd_pkts=5 ogf=0x4 ocf=0x1 status=0 hci_ver=8 hci_rev=782 lmp_ver=8 mfrg=96 lmp_subver=782
17:24:25.729 -> ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0003 len=0
17:24:25.729 -> 0x03 0x10 0x00 
17:24:25.729 -> Command complete: cmd_pkts=5 ogf=0x4 ocf=0x3 status=0 supp_feat=0x877bffdbfecdeebf
17:24:25.729 -> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0001 len=8
17:24:25.764 -> 0x01 0x0c 0x08 0x90 0x80 0x00 0x02 0x00 0x80 0x00 0x20 
17:24:25.764 -> Command complete: cmd_pkts=5 ogf=0x3 ocf=0x1 status=0 
17:24:25.764 -> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0063 len=8
17:24:25.764 -> 0x63 0x0c 0x08 0x00 0x00 0x80 0x00 0x00 0x00 0x00 0x00 
17:24:25.764 -> Command complete: cmd_pkts=5 ogf=0x3 ocf=0x63 status=0 
17:24:25.764 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0001 len=8
17:24:25.764 -> 0x01 0x20 0x08 0x7f 0x06 0x00 0x00 0x00 0x00 0x00 0x00 
17:24:25.764 -> Command complete: cmd_pkts=5 ogf=0x8 ocf=0x1 status=0 
17:24:25.764 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0002 len=0
17:24:25.764 -> 0x02 0x20 0x00 
17:24:25.834 -> Command complete: cmd_pkts=5 ogf=0x8 ocf=0x2 status=0 
17:24:25.834 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0003 len=0
17:24:25.834 -> 0x03 0x20 0x00 
17:24:25.834 -> Command complete: cmd_pkts=5 ogf=0x8 ocf=0x3 status=0 
17:24:25.834 -> ble_hs_hci_cmd_send: ogf=0x04 ocf=0x0009 len=0
17:24:25.834 -> 0x09 0x10 0x00 
17:24:25.834 -> Command complete: cmd_pkts=5 ogf=0x4 ocf=0x9 status=0 bd_addr=30:ae:a4:5:e7:a
17:24:25.834 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0018 len=0
17:24:25.834 -> 0x18 0x20 0x00 
17:24:25.834 -> Command complete: cmd_pkts=5 ogf=0x8 ocf=0x18 status=0 rand=0xa74f169328b4e5fc
17:24:25.834 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0018 len=0
17:24:25.869 -> 0x18 0x20 0x00 
17:24:25.869 -> Command complete: cmd_pkts=5 ogf=0x8 ocf=0x18 status=0 rand=0x9efd246861bceebc
17:24:25.869 -> Device added to RL, Resolving list count = 1
17:24:25.869 -> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0031 len=1
17:24:25.869 -> 0x31 0x0c 0x01 0x01 
17:24:25.869 -> Command complete: cmd_pkts=5 ogf=0x3 ocf=0x31 status=0 
17:24:25.869 -> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0033 len=7
17:24:25.869 -> 0x33 0x0c 0x07 0xff 0x00 0x00 0x0c 0x00 0x00 0x00 
17:24:25.869 -> Command complete: cmd_pkts=5 ogf=0x3 ocf=0x33 status=17 
17:24:25.869 -> ble_hs_hci_cmd_send: ogf=0x03 ocf=0x0031 len=1
17:24:25.869 -> 0x31 0x0c 0x01 0x00 
17:24:25.869 -> Command complete: cmd_pkts=5 ogf=0x3 ocf=0x31 status=0 
17:24:25.904 -> looking up peer sec; 
[... 64 lines cut ...]
17:24:26.147 ->          handle 10
17:24:26.147 ->      end_handle 13
17:24:26.147 -> characteristic
17:24:26.147 ->            uuid 00001624-1212-efde-1623-785feabcd123
17:24:26.147 ->      def_handle 11
17:24:26.147 ->      val_handle 12
17:24:26.147 ->    min_key_size 0
17:24:26.147 ->           flags [READ|WRITE_NO_RSP|WRITE|NOTIFY]
17:24:26.147 -> ccc descriptor
17:24:26.147 ->            uuid 0x2902
17:24:26.147 ->          handle 13
17:24:26.147 ->    min_key_size 0
17:24:26.147 ->           flags [READ|WRITE]
17:24:26.147 -> GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=32 adv_itvl_max=64
17:24:26.182 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0006 len=15
17:24:26.182 -> 0x06 0x20 0x0f 0x20 0x00 0x40 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x07 0x00 
17:24:26.182 -> Command complete: cmd_pkts=5 ogf=0x8 ocf=0x6 status=0 
17:24:26.182 -> ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000a len=1
17:24:26.182 -> 0x0a 0x20 0x01 0x01 
17:24:26.182 -> Command complete: cmd_pkts=5 ogf=0x8 ocf=0xa status=0 
17:24:26.182 -> D NimBLEAdvertising: "<< Advertising start"
17:24:26.182 -> [D][Lpf2HubEmulation.cpp:447] start(): Characteristic defined! Now you can connect with your PoweredUp App!
```

No I need to thank you for all the effort ;)

corneliusmunz, 2020-10-23:
@AlbanT i have tried it on two phones. One with Android 6.0.1 and one with Android 10. On both the HubEmulation.ino sketch worked fine. It is very strange, that after starting the Advertisment of the Bluetooth Hub (last log message) nothing happens. Normally if the app finds a device which has the correct advertisment values it initiates a connect and requests several other messages. 

I have to search in the NimBLE-Arduino issues if someone else has experienced the same issue on Android devices. 

Maybe you can install a Bluetooth Scanner on your Android device and try to make a screenshot of the Bluetooth advertisment data (https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp)

AlbanT, 2020-10-23:
Hi @corneliusmunz is this what you need?

'''
nRF Connect, 2020-10-23
Hub (30:AE:A4:05:E7:0A)
V	18:31:09.015	Connecting to 30:AE:A4:05:E7:0A...
D	18:31:09.015	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE)
D	18:31:09.668	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
D	18:31:09.714	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	18:31:09.714	Connected to 30:AE:A4:05:E7:0A
V	18:31:09.749	Discovering services...
D	18:31:09.749	gatt.discoverServices()
D	18:31:10.589	[Callback] Services discovered with status: 0
I	18:31:10.589	Services discovered
V	18:31:10.619	Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Unknown Service (00001623-1212-efde-1623-785feabcd123)
- Unknown Characteristic [N R W WNR] (00001624-1212-efde-1623-785feabcd123)
   Client Characteristic Configuration (0x2902)
D	18:31:10.620	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	18:31:10.625	gatt.setCharacteristicNotification(00001624-1212-efde-1623-785feabcd123, true)
'''

AlbanT, 2020-10-23:
Raw data:

`0x09FF97030041071D6300110723D1BCEA5F782316DEEF1212231600000409487562020106020A00051210002000`

corneliusmunz, 2020-10-27:
Hi @AlbanT ! Unfortunately i was not able to find the root cause for the behaviour on your phone. The advertisment data looks good for me but the connection procedure does not start. I have really no glue why this happens 😢

AlbanT, 2020-10-27:
Hi @corneliusmunz pity :(
Thanks for the effort!!!!
I'll have a go with some other android phones whenever I come across one :)

hrgraf, 2020-11-01:
Hi,
just want to let you know, that the HubEmulation example does not work with my android phone (Android 9) neither.
While I can connect with the "nRF Connect tool", the Lego Powered Up app is unable to connect, but somehow reports at least the advertised name (TrainHub). I have enabled both debug reportings, but no message happens after
```
D NimBLEAdvertising: "<< Advertising start"
[D][Lpf2HubEmulation.cpp:447] start(): Characteristic defined! Now you can connect with your PoweredUp App!
```

Otherwise, I can connect with my Android phone to my Lego Boost Hub and control it from the PoweredUp app e.g. as batmobile.

corneliusmunz, 2020-11-01:
Hi @hrgraf, @AlbanT ! Maybe it has something to do with the NimBLE-Arduino library. I will contact the owner of the lib if he can guide us in some direction

hrgraf, 2020-11-01:
Hi @corneliusmunz, @AlbanT,

I just did another try, and now I was able to connect with Powered Up to the HubEmulation on my Android 9 phone.
This time, I did not only kill the Powered Up app, but also disable and re-enable the Bluetooth radio in the phone.
Or was I just lucky?

Once connected, I succeed in reconnecting. May-be the Powered Up app keeps some connection cache?
Seems like, as it has an auto-connect option in the settings.
When starting the app, it does now auto-connect to the HubEmulation within 1s.

corneliusmunz, 2020-11-01:
@hrgraf many thanks for reporting this observation. @AlbanT Can you try also to kill the PoweredUp App and disable and re-enable the Bluetooth radio in the phone?

AlbanT, 2020-11-02:
Hi @corneliusmunz I think @hrgraf nailed it.
I can now connect after killing both the app and the bluetooth on my tablet and re-enabling the BT and restarting the app the connection was made succesfully.

closed 2020-11-02 by AlbanT

corneliusmunz, 2020-11-02:
@AlbanT I am really happy that it worked on your side! Thank you for checking it again. Unfortunately this is an issue which is not controllable from the library side. Maybe it helps if the Device Address changes every time. But i am not sure. I will do a short check.  
@hrgraf Thank you again for figuring out how we can overcome the conntection issues!

wmarkow, 2020-12-06:
Hi guys, I also have some issues with emulated hub and Powered Up in Android. I already put some comments in https://github.com/corneliusmunz/legoino/issues/12 but here it is a better place to comment.

Basically emulated hub for me doesn't work with Powered on Up Android; it didn;t want to connect. I need to make this trick to close the app, clear its application data etc. It works until I connect to a real Lego hub. After this the emulated hub doesn't connect, which is a pain. Good thing: somehow I managed to make it working! It works. I will just comment what I did and come back to this later.

What I did. I have installed [BLE Sniffer](https://play.google.com/store/apps/details?id=com.aconno.blesniffer) application which display the advertising data as a hex dump.

For a real Lego Hub I have the following (all in encoded as HEX):
```
02 01 06 11 07 23 d1 bc
ea 5f 78 23 16 de ef 12
12 23 16 00 00 09 ff 97
03 00 41 07 b1 43 00 05
12 10 00 20 00 02 0a 00
0a 09 53 6d 61 72 74 20
48 75 62
```

which gives us the folling advertising data in this specific order:
```
02 01 06                                              length=02 type=01 (flags) value=06
11 07 23 d1 bc ea 5f 78 23 16 de ef 12 12 23 16 00 00 length=11 type=07         value=23 d1...
09 ff 97 03 00 41 07 b1 43 00                         length=09 type=ff         value=97 03... Lego Manufacturer Data
05 12 10 00 20 00                                     length=05 type=12
02 0a 00                                              length=02 type=0a
0a 09 53 6d 61 72 74 20 48 75 62                      length=0a type=09         value=53 6d... Smart Hub
```

For my emulated hub I have the folowing advertising data:
```
09 ff 97 03 00 41 07 1d
63 00 11 07 23 d1 bc ea
5f 78 23 16 de ef 12 12
23 16 00 00 04 09 48 75
62 02 01 06 02 0a 00 05
12 10 00 20 00
```
which gives us the folling advertising data in this specific order:
```
09 ff 97 03 00 41 07 1d 63 00                        length=09 type=ff         value=97 03... Lego Manufacturer Data
11 07 23 d1 bc ea 5f 78 23 16 de ef 12 12 23 16 00 00 length=11 type=07         value=23 d1...
04 09 48 75 62
02 01 06
02 0a 00
05 12 10 00 20 00
```
 
It is almost the same as the data from original Lego Hub. The only differences I noticed are:
 * three last bytes of manufacturer data are different
 * the order of data packets is different

I have changed the last three bytes of manufacturer data in **Lpf2HubEmulation::start()** to be exactly the same as in the real Lego Hub. It didn't help. Then I had this tought: it is impossible that the order of data sent in the advertisement could not have an impact here. Probably it has! I have reworked a bit **Lpf2HubEmulation::start()** method and now it works. The emulated hub connects to Powered Up. it works now in 95%.

In **Lpf2HubEmulation::start()** I did as below:
```
  // wmarkow begin disable scan response
  //  _pAdvertising->setScanResponse(true);
  _pAdvertising->setScanResponse(false);
  // wmarkow end
```
and
```
  // wmarkow begin: set custom order of advertisement data
  //_pAdvertising->setAdvertisementData(advertisementData);
  //_pAdvertising->setScanResponseData(scanResponseData);
  NimBLEAdvertisementData wmarkowAdvertisementData = NimBLEAdvertisementData();
  wmarkowAdvertisementData.addData(std::string{0x02, 0x01, 0x06});
  wmarkowAdvertisementData.addData(std::string{0x11, 0x07, 0x23, 0xd1, 0xbc, 0xea, 0x5f, 0x78, 0x23, 0x16, 0xde, 0xef, 0x12, 0x12, 0x23, 0x16, 0x00, 0x00});
  wmarkowAdvertisementData.addData(std::string{0x09, 0xff, 0x97, 0x03, 0x00, 0x41, 0x07, 0xb1, 0x43, 0x00});
  wmarkowAdvertisementData.addData(std::string{0x05, 0x12, 0x10, 0x00, 0x20, 0x00});
  wmarkowAdvertisementData.addData(std::string{0x02, 0x0a, 0x00});
  wmarkowAdvertisementData.addData(std::string{0x0a, 0x09, 0x3, 0x6d, 0x61, 0x72, 0x74, 0x20, 0x48, 0x75, 0x62});
  _pAdvertising->setAdvertisementData(wmarkowAdvertisementData);
  // wmarkow end
```
After this change the hub connects to Powered Up, however **BLE Scanner** shows that not all data are sent (I do not know why). Only this data appear in advertisement:
```
  wmarkowAdvertisementData.addData(std::string{0x02, 0x01, 0x06});
  wmarkowAdvertisementData.addData(std::string{0x11, 0x07, 0x23, 0xd1, 0xbc, 0xea, 0x5f, 0x78, 0x23, 0x16, 0xde, 0xef, 0x12, 0x12, 0x23, 0x16, 0x00, 0x00});
  wmarkowAdvertisementData.addData(std::string{0x09, 0xff, 0x97, 0x03, 0x00, 0x41, 0x07, 0xb1, 0x43, 0x00});
```

@corneliusmunz I can create my fork of your repo, create some custom branch and commit my minimal changes that makes it working with Powered Up on Android because for now - I think - it is hard to cooperate and I can introduce here a mess. Then we can find out the following:
 * why this change makes it working on Android? Maybe it works becaus I have disabled scan response with ```_pAdvertising->setScanResponse(false);```?
 * does it have a negative impact on iPhone Lego application