corneliusmunz/legoino #29: Legoino 1.0.1 do not compile on arduino IDE 1.8.3

Gcopper1984, 2020-10-27:
Hello,
 
   I just installed Legoino library for arduino 1.8.3 but the  "DuploTrainExample" do not compile (here the error log file). I edit the "Lpf2Hub.cpp" file (where the error were reported) and modified line 106 (error in implicit cast), line 784 (return is missing, I put -1 but unsure of the result) and line 466 (in for loop index i is not initialized).

With new file the example compile, but I am not sure about results.

Thanks,

GB

corneliusmunz, 2020-10-27:
Hi @Gcopper1984 ! Thanks for opening your issue. The return value is definitely missing. Thanks for reporting this. I will correct that in the next version.

About the compiler error: This should work and i have no issues in compiling and running the example. Can you please try to update your Arduino version to the current one (1.8.13) 1.8.3 is from 2017 and maybe there are some changes. Can you provide me the error log file?

Gcopper1984, 2020-10-28:
Good morning,

   ok, I upgraded to Arduino 1.8.13 and I also re-install library from library manager but error in compiling still there. Here the log file from Arduino IDE

[LegoinoError.txt](https://github.com/corneliusmunz/legoino/files/5450345/LegoinoError.txt)
 

Anyway, great work! I am waiting for my esp32 board to test, I hope to do some real "test" :-) very soon.

Have a nice day and thank you one more time.

corneliusmunz, 2020-10-29:
I have removed all warnings and will provide a bugfix Version soon

corneliusmunz, 2020-11-04:
fixed with release https://github.com/corneliusmunz/legoino/releases/tag/1.0.2

closed 2020-11-04 by corneliusmunz