corneliusmunz/legoino #30: Power Functions Sample Connection

Hantse, 2020-10-28:
Hello,

Very nice library, i like, it's really funny and usefull.

But, it's possible to have a full connetion scheme with IR for Power Functions ?

I try with IR led on Pin12, but for me, Power Functions doesn't work, probably my fault, in electronic part ?

Kind regards,

Gcopper1984, 2020-10-29:
Hello,

   for a cheap and dirty test you can try to connect your led directly to the output of the board in series with a 220 ohm for limiting source current from your IC but do not expect great working distance. At first try place the led near the lego receiver.

Take this as example:
https://create.arduino.cc/projecthub/karmette/basic-led-setup-for-beginners-0a124a

You have to set a lower resistor value because esp32 board have a 3.3V output voltage level. 

To get more working distance you have put a transistor as a "switch" for your led to get more current flow from it, here another example:

https://create.arduino.cc/projecthub/sumanbargavr/switching-using-transistor-c9114a

Three more thing:

1) in power function library the "channel" value is the value of the channel of your receiver - 1;
2) IR leds, in general, have different wavelenght and maybe your could not match (and so poor performance) the wavelenght of the lego IR receiver.  
3) IR leds have also a "cone" light projection shape. Maybe you have to put your led with a good 
alignment with the receiver to get it to work properly.

Hope this help. 

GC84

corneliusmunz, 2020-10-29:
Hi @Hantse! I have tried it with the direct connection of a IR Led with an 220 ohm resistor as proposed by @Gcopper1984. What sketch do you have used? Was it the example sketch which is part of the library? As @Gcopper1984 already said: If you have selected channel 1 on your Powerfunctions IR receiver, you have to use channel 0 on the library. Maybe i will change that in the future to have the same numbers like the IR receiver is using. 

Here you can find an example sketch and a schematic:
[image]
[image]

Hantse, 2020-11-04:
Hello ! 

Thanks for your feedback guys, all work fine for me :D

closed 2020-11-04 by corneliusmunz