corneliusmunz/legoino #31: Duplo train - Invalid color name for no-color

Gcopper1984, 2020-10-29:
Hello once again.

   I try the DuploTrainBase example and work properly with my board.

But I noticed  that if I lift the train base the callback function report a color of (int) 255 so in the serial display appear an invalid name (something that is readed like "???").

Could be useful add another color named "nocolor" (for example) in the COLOR_STRING array or map the color id > 10 as "black" in the parseColor() function.

Thank you,

GC84

corneliusmunz, 2020-10-29:
Hi @Gcopper1984 ! Very good finding! I will fix this and add an additional Color in the Color array

corneliusmunz, 2020-11-02:
Will be fixed together with your proposed change in the next released version 1.0.3. I have introduced now a new entry in the COLOR_STRING array and added a static funtion which handles the conversion from color/int to a string. This method will handle all color values above NUM_COLORS as "none".

corneliusmunz, 2020-11-04:
fixed with release https://github.com/corneliusmunz/legoino/releases/tag/1.0.2

closed 2020-11-04 by corneliusmunz