corneliusmunz/legoino #35: Read reply from HUB

Gcopper1984, 2020-11-02:
Hi @corneliusmunz,

   thank you for your support, is a great thing that this project is alive!

Another question: I am digging into BLE protocol and I there are some command where HUB send reply to the Arduino. 
For example command "Port Information Request" or "Port Input Format Setup"

So, if I understend correctly, I can send message from arduino to the HUB with the "WriteValue()" command. But how can I get access of this reply? There is a callback to register?

Or, there is already a function that handle reply for theese commands?

Thank you

corneliusmunz, 2020-11-02:
Hi @Gcopper1984! Yes, you can request a port Information or Port Mode Information request via the WriteValue command. But currently the callback function is not implemented for the Port Information and Port Mode Information messages. But you can change/try it by your own in changing the implementation of the common `notifyCallback` method inside the Lpf2Hub.cpp file. This callback is fired by the NimBLE-Arduino library when a value on the characteristic has changed. The code should be adapted as follows:

```c++
void Lpf2Hub::notifyCallback(
    NimBLERemoteCharacteristic *pBLERemoteCharacteristic,
    uint8_t *pData,
    size_t length,
    bool isNotify)
{
    log_d("notify callback for characteristic %s", pBLERemoteCharacteristic->getUUID().toString().c_str());

    switch (pData[2])
    {
    case (byte)MessageType::HUB_PROPERTIES:
    {
        parseDeviceInfo(pData);
        break;
    }
    case (byte)MessageType::HUB_ATTACHED_IO:
    {
        parsePortMessage(pData);
        break;
    }
    case (byte)MessageType::PORT_VALUE_SINGLE:
    {
        parseSensorMessage(pData);
        break;
    }
    case (byte)MessageType::PORT_OUTPUT_COMMAND_FEEDBACK:
    {
        parsePortAction(pData);
        break;
    }
    case (byte)MessageType::PORT_INFORMATION:
    {
        // do some stuff here
        break;
    }
    case (byte)MessageType::PORT_MODE_INFORMATION:
    {
        // do some stuff here
        break;
    }
    }
}
```

So in the last two cases you can do some stuff you want if the message about the port is received. I have implemented it the other way round in the HubEmulation.cpp file. In this case the PoweredUp App requests port information and port mode information values and i have to reply with some meaningful values to "mock" the attached device on a specific port. You can see some sequence diagramms in the LEGO wireless proctocoll specification:
https://lego.github.io/lego-ble-wireless-protocol-docs/index.html#sequence-diagrams

Hope this helps!

If you want to contribute with a Pull request, i would be totally happy! The project lives because some users contribute in raising issues, asking questions, telling use-cases which i have never thought of... so overall thanks for your contribution 👍

Gcopper1984, 2020-11-03:
Good morning,

   ok, I will try something for sure about callback, many thanks for the indication. and if I will be able to do somenthing good I will try with a Pull request.

Have a nice day!

Gcopper1984, 2020-11-03:
Hi,

   I think I got the point and start implementation. I also added virtual port command and create a pull request.

@Gcopper1984

corneliusmunz, 2020-11-04:
Hi @Gcopper1984 ! Thanks for your work an PR. I will have a look on it the next days.

Cheers,
Cornelius