corneliusmunz/legoino #37: HubEmulation support for more devices

hrgraf, 2020-11-04:
Hi @corneliusmunz,

the HubEmulation is a great feature for the legoino library and will open tons of new applications, e.g. to control ESP32 projects with the Lego PoweredUp app.

While basic motor drive commands arrive, drive for angle etc. seems not be supported (yet)?
And what about other sensor data? Having bidirectional communication would increase possibilities.

I have started a [HubEmulation example](https://github.com/hrgraf/M5) to emulate a Boost Move Hub using
an ESP32 Atom Matrix and try reporting its accelerometer as tilt to the PoweredUp app.

In legoino V1.0.2, Lpf2HubEmulation::start() does only support HubType::POWERED_UP_HUB resp. CONTROL_PLUS_HUB.
So in my fork, I have added manufacturerData for the BOOST_MOVE_HUB.

Next, getPortInformationPayload() and getPortModeInformationRequestPayload() support only DeviceType::TRAIN_MOTOR and HUB_LED. How to add more devices, e.g. MOVE_HUB_TILT_SENSOR?

Where/how did you get the payload values in those functions?

My approach is very tedious and takes many iterations: I use another sketch with a legoino Boost hub, where I request the same from the real Lego Boost Hub and log the answers. Then, I can enter them in the HubEmulation and have another try.

Any better idea or more documentation on this?
Any plans on such features on your side?

corneliusmunz, 2020-11-04:
Hi @hrgraf !

Thanks for your feedback! I am really happy that this beta feature seems usefull for you. Many thanks for your contributions so far!

Now to your issue:

Good Idea to use the Accelerometer of the Atom matrix as input device for the HubEmulation. I initially want to cover only a few use cases to see if the community will get the idea and requests more features. Now this ist the case and i can implement the Move hub and could also try to implement other devices. Do you have a priority for me?

1) MoveHub 
2) TiltSensor for MoveHub and ControlPlusHub
3) ??

Maybe we do it step by step together on a feature branch and you can test it iteratively. 

I got the data of the hubs and sensors from the following .net c# library where you can fetch hub information and device information. The library has a good CLI where the Mode and Port information values are requested and logged:
https://github.com/sharpbrick/powered-up

The maintainer of the library @tthiery has also started a documentation project where you can find the responses and modes of the different devices:
https://github.com/sharpbrick/docs

It is definitely not complete but a good starting point and with the CLI commands described here you can fetch the most of the needed information which is used to enhance the HubEmulation project
```
Command Line (dotnet tool install -g SharpBrick.PoweredUp.Cli)
 poweredup device list (discover all connected devices and their port (mode) properties)
 poweredup device dump-static-port -p <port number> (see adding new devices tutorial)
```

Cheers,
Cornelius

hrgraf, 2020-11-05:
Hi @corneliusmunz,

thank you for the links and information.

Yes, my priorities are
1. MoveHub
2. TiltSensor
3. TachoMotor

I have started implementing the first 2 tasks, by adding manufacturerData, portInformation, portModeInformation.
Unfortunately, the AtomMatrix emulating the hub is crashing with the following message:

Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (ble) 

I have tried to add some delay() to not overload the BLE communication, but did not help.

Have you ever run into such issues?
I have no experience with the nimble stack. 
Do you know how to increase its memory configuration / stack size?

I might also try with another ESP32 board with more resources (not pico).

Cheers,
HR

hrgraf, 2020-11-05:
OK, I have tweaked some constants in nimconfig.h to increase stack and buffers.
Now, the ESP32 does not crash anymore.

hrgraf, 2020-11-05:
OK, PoweredUp app connects on 2nd attempt. So I probably just need so send tilt data

WilixLead, 2020-11-13:
Hello guys! First of all - it is a great Idea and I working on emulation device some time before https://wibrick.wilix.org/
Not I just don't have needed time for it. But I still want create smaller and custom controller with LiPo battery.
Anyway. My question about Distance, Color and Force sensors. Some one have captured protocol packets for one of this devices? I just want implement some own (like Force sensor - just a button)

I have needed hardware for capture, but don't have any sensor for capture :)

AlbanT, 2020-11-29:
Can I suggest compatibility for the colorsensor.
Either the Lego colorsensor (although getting hold on female connectors for these devices is not easy, any suggestions?) or for example a command that we can feed our own data so it is reconized by the get color functions in the app.
I'm planning to try out the M5Stack TCS34725 Sensor Module.

corneliusmunz, 2020-11-29:
Hi @AlbanT , @WilixLead I am currently working on the support of additional hubs and sensors for the hub emulation. A good source of the sensor messages like supported modes the following project is a good source for protocol documentation of concrete sensors: https://github.com/sharpbrick/docs But be a littlebit patient for the additional devices.. maybe it will become a christmas gift 😉

tthiery, 2020-12-01:
Regards https://github.com/sharpbrick/docs ... 

Happy to accept contribution. I have seen such a disparity in documentation, so I hoped to pull some ends together.

I have the Mindstorm set by now but have not "processed" the goodies yet. It is unfortunately not only pulling the metadata from the devices (which is an important part) but also the deviations (like: Oops, wrong metadata) and dirty details of each sensor (like: the Mario's rgb sensor works only if you do not consume the tag sensor at the same time despite it is an allowed combined mode).

@WilixLead I can provide you short term the metadata for color and the distance sensor. I do not have the force sensor (the one with the Spark Prime set). Maybe Nathan from the JS Lib can help us creating a trace. I think he has them.

WilixLead, 2020-12-01:
> Regards https://github.com/sharpbrick/docs ...
>
> Happy to accept contribution. I have seen such a disparity in documentation, so I hoped to pull some ends together.
>
> I have the Mindstorm set by now [...]

Thank you! Unfortunetly I still not done my schematic for WiBrick controller, it takes too much time at this moment. I hope do it in current month...
If some one can help me with schema and pcb (I want compact and powerful device) - welcome (fully opensource :))

jcmeier, 2020-12-19:
Hello, 

thanks for your efforts so far. I played around a bit with the Hub feature and the PoweredUp app on Ios. It works quite well, but   did I understand it is a bit limited at the current point?

The train controls works quite fine, and I can see then in the serial console of the ESP 32. But a freely created control I do not get up and running.

tthiery, 2020-12-21:
@WilixLead I added the traces for the Color Sensor and the Distance sensor to [sharpbrick/docs project](https://github.com/sharpbrick/docs). However, I have not analyzed them yet.

corneliusmunz, 2020-12-21:
@jcmeier Currently the feature is more like a "beta" feature and supports only the PoweredUp Hub (Train) and the lego technic hub. Could you describe the control you want to implement and what are in detail the problems with your sketch? Maybe i can help you

d0m1n1qu3, 2020-12-30:
Is it possible to controll the Emulated Hub with the LEGO Train WirelessController? that would be cool for me 😎👍🏻

realhuno, 2024-03-06:
> Is it possible to controll the Emulated Hub with the LEGO Train WirelessController? that would be cool for me 😎👍🏻

Yes i also looking for this. Is it possible to controll the Emulated Hub (with PF) with the original Train Wireless Controller?