corneliusmunz/legoino #39: Feature Request HubEmulation: Support for regulated Motors

StefansProjects, 2020-12-01:
Thank you for your project, its really great and fun to work with!

Currently I am working on a ESP32-based legoino hub to connect old NXT / EV3 motors (and eventually NXT / EVE3 sensor) to the new PoweredUp app (and other devices using the new protocol).  Hardware is rather simple, but unfortunately it seems like its currently not possible to register anything but train motors, or is it?

Also the callback  only seems to support one byte channel and one byte payload. No subcommands nor multibyte payloads seem to be supported, which are necessary for more advanced control for regulated motors.

So can you give me a hint whether you are working on these problems or where to look myself to provide a patch?

corneliusmunz, 2020-12-01:
Hi @StefansProjects ! Thanks for your feedback 😄 The hub emulation part was some kind of beta feature in the 1.0.0 version of the library to see if the users need it. I now get more and more feedback and feature requests and i think it is worth to do some more work on that feature. I am currently working on a feature branch on the following topics:

* support MoveHub (aka BoostHub)
* Support of sensor feedback to have methods like `setTiltSensorValues(port, x, y, z)` , `setColorDistanceSensorValues(port, color, distance)` , setMotorAngle(port, absoluteAngle)` 
* Automatic feedback of port information and port mode information for different sensor values

I am facing some issues about the sensor feedback but i think i am on a good way. It will take some time but i think till Christmas the additional features could be available in a new version of the library. As soon as the feature branch has some kind of stable state, i will push it to the repo to get feedback from you before i will make a new release. 

Currently you could already write "manually" more than one byte payload to the characteristic value in using a code snipped as the followin one:

```c++
  std::string payload = "";
  payload.push_back((char)port);
  payload.push_back((char)Event::ATTACHED_IO);
  payload.push_back((char)deviceType);
  std::string versionInformation = {0x00, 0x00, 0x00, 0x00, 0x10, 0x00, 0x00, 0x00, 0x10};
  payload.append(versionInformation); //version numbers
  writeValue(MessageType::HUB_ATTACHED_IO, payload);
```

Cheers,
Cornelius

StefansProjects, 2020-12-04:
Looking forward for your additions. Building the hardware meanwhile