corneliusmunz/legoino #45: Add support for Mario Hub (#71360)

corneliusmunz, 2020-12-21:
Add support for Mario Hub (set number #71360) 

* Gesture Sensor
* Pants Sensor
* Color/Barcode Sensor

Source of knowledge: 
* https://github.com/bricklife/LEGO-Mario-Reveng
* https://github.com/sharpbrick/powered-up
* https://github.com/sharpbrick/powered-up/issues/91

corneliusmunz, 2020-12-21:
The new features are implemented on the develop branch in the commit https://github.com/corneliusmunz/legoino/commit/3d537c6a70a5c008bb7f82627bca6e3f5b065fc0 

Would be great if someone could test the example `Mario.ino` sketch 😄

corneliusmunz, 2020-12-22:
Fixed with release https://github.com/corneliusmunz/legoino/releases/tag/1.1.0

corneliusmunz, 2020-12-22:
@tthiery, @bricklife I have seen in the app, that you can control the volume of the mario loudspeaker but i found no "volume" hub property or port. I found during the connection procedure, that a HubProperty with reference code `0x12` was requested and the hub answered with `0x06, 0x00, 0x01, 0x12, 0x06, 0x64` In that case the Mario was set to 100% volume via the app and maybe `0x64` could be the volume value (DEC `100`) Have you done some evaluations on this? Or maybe the volume is part of the debug Port `3`

bricklife, 2020-12-22:
Yes, `0x12` is the undocumented property reference of volume. You can set a value (as volume %) with a command `06 00 01 12 01 xx`.

corneliusmunz, 2020-12-22:
@bricklife cool! Thanks for your fast feedback. Then i will implement it in the library