corneliusmunz/legoino #46: Unable to connect multiple hub since release 1.0.4

Mattze0815, 2020-12-22:
Hi @corneliusmunz,
we appreciate the non-blocking connection procedure introduced with release 1.0.4. This is a great improvement, thank you very much for that.

We have experienced the following problem:
- We want to connect to multiple hubs.
- You call init() for multiple hubs BEFORE the hubs actually connect to the PC.
- Only the last hub that was initialised will be connected, not the other ones.

We had to create a relatively complitcated work around to overcome this problem.

Maybe you could have a look into this?

Thanks!

Mattze0815, 2020-12-22:
FYI;: problem also exists in 1.1.0.

corneliusmunz, 2020-12-22:
Hi @Mattze0815 ! Yes.. unfortunately that behaviour is strictly related to the "non blocking" behaviour. I think that is good but also "evil" I have discussed this with @fvanroie: https://github.com/corneliusmunz/legoino/issues/41#issuecomment-747098217 I think i will revert the non blocking behaviour because it needs a very complicated logic to detect if scans are already ongoing and unfortunately if a scan is ongoing, a hub could not be connected at the same time. This is related to a constraint in the NimBLE stack : https://github.com/h2zero/NimBLE-Arduino/issues/58#issuecomment-632669156

corneliusmunz, 2020-12-22:
@Mattze0815 Maybe i will add a optional parameter to enforce non-blocking scans but not as a default behaviour. I am currently working on a `HubManager` which will have the logic and boilerplate inside if you want to have a system with different hubs: https://github.com/corneliusmunz/legoino/issues/41#issuecomment-747554701

fvanroie, 2020-12-23:
I can make a post of my multi-hub project. The code is not pretty but it works with both scan methods.
Or if you add me to the mattzo-controller repo, I'll have a look!

corneliusmunz, 2020-12-23:
@fvanroie would be great if you could send me your working example or post it here.

Mattze0815, 2020-12-23:
Hi guys, your answers are much appreciated, thank you.

I would very strongly wish to keep the non-blocking connectivity feature, even wish the present limitation. I have build a work around, and it appears to work quite stable. It's just not very elegant.

@fvanroie, I will add you to MattzoBricks shortly and give you some instructions of were to look. Thanks in advance!

fvanroie, 2020-12-23:
I have uploaded my [PUp TrainController](https://github.com/fvanroie/PUp-TrainController) project as a github repo. The main interest is in [`ble_setup()`](https://github.com/fvanroie/PUp-TrainController/blob/5ec6be6f0b1c82e0ecb0ecbb2b4d19d73ad5d739/src/lego_ble.cpp#L499) and [`ble_hub_task()`](https://github.com/fvanroie/PUp-TrainController/blob/5ec6be6f0b1c82e0ecb0ecbb2b4d19d73ad5d739/src/lego_ble.cpp#L299) functions.

Mattze0815, 2020-12-23:
@fvanroie, access granted.

Look for the MattzoController/MTC4PU folder. You need to copy the MattzoController/libraries/MattzoBricks folder to your Arduino library folder.

Connecting multiple hubs is possible (successfully tested with 8 concurrently connected hubs), but they need to be connected exactly in the order as specified in MTC4PU_Configuration.h.

Enjoy the fun: https://github.com/Mattze0815/roc2bricks

More general info here: www.mattzobricks.com