corneliusmunz/legoino #47: Callback functions for connection status

Mattze0815, 2020-12-23:
Hi @corneliusmunz,
would it be able to introduce some handy callbacks for the following events?

1.
connecting(): hub is connecting

2.
connected(): hub has been connected. Might only be useful if .connect() is converted into a non-blocking function as well.

3.
disconnected(): hub has disconnected

Keep up to good work!

fvanroie, 2020-12-23:
Indeed I think these callbacks will be very handy in a simplified flow. I proposed these as well in [another comment proposal](https://github.com/corneliusmunz/legoino/issues/41#issuecomment-747677785) for the HubManager feature. This was in relation to *Make connection flow easier #38*.

I haven't looked at the new HubManager class yet, but I think these callbacks would align with that new feature...

corneliusmunz, 2020-12-23:
@fvanroie @Mattze0815 I will try to consider this feature request in the work on the HubManager class. Unfortunately i found not so much time till now but during the holidays i will try to tackle this 😄