corneliusmunz/legoino #50: HUB emulation PORT A

smotek, 2020-12-29:
Hi,
legoino is a perfect project, thank you.
I have a problem with HUB emulation.
HW I use NodeMcu WROOM-32
I monitor the data on a serial monitor.
Everything is OK for PORT B.
For HUB LED it is OK.
But I don't get any data for PORT A.
Who could be the problem?
Will emulation also be available for TechnicHUB?
Thank you

corneliusmunz, 2021-01-06:
Hi @smotek ! First the emulation should work for the TechnicHUB and the TrainHUB but right now only for the train motors. I am developing a new version where more motors and also sensors are supported. Regarding your problem of port A. Maybe you have to increase some delays a littlebit. I sometimes see that the command/response messages was not exchanged correclty and the app then does not recognize the "attached" device on port A. If so, no message will be send out of the app even if you have a block which addresses port A

smotek, 2021-01-06:
Hi corneliusmunz thanks for the info.
I probably found it in the code Type Hub I should use CONTROL_PLUS_HUB
and add rows to the loop for C and D.

To fix PORT A, I should try to change the times in the loop
port detection? (1000 ms)
Thank you

corneliusmunz, 2021-01-06:
Hi @smotek yes... you should initialize the object like the following 
```c++
// create a hub instance
Lpf2HubEmulation myEmulatedHub("TrainHub", HubType::CONTROL_PLUS_HUB);
```

And for the Port A, just try to change the delays in the following section (decrease it and increase it):
```c++
    delay(1000);
    myEmulatedHub.isPortInitialized = true;
    myEmulatedHub.attachDevice((byte)PoweredUpHubPort::A, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub.attachDevice((byte)PoweredUpHubPort::LED, DeviceType::HUB_LED);
    delay(1000);
    myEmulatedHub.attachDevice((byte)PoweredUpHubPort::B, DeviceType::TRAIN_MOTOR);
    delay(1000);
```

smotek, 2021-01-06:
If I use
Lpf2HubEmulation myEmulatedHub ("TrainHub", HubType :: CONTROL_PLUS_HUB);
reports error for rows
myEmulatedHub.attachDevice ((byte) PoweredUpHubPort :: C, DeviceType :: TRAIN_MOTOR);
     delay (1000);
myEmulatedHub.attachDevice ((byte) PoweredUpHubPort :: D, DeviceType :: TRAIN_MOTOR);
     delay (1000);
     
    error: 'C' is not a member of 'PoweredUpHubPort'
    
delay for A test

smotek, 2021-01-06:
PORT A  solved

 myEmulatedHub.isPortInitialized = true;
     delay(1000);
 myEmulatedHub.attachDevice((byte)PoweredUpHubPort::A, DeviceType::TRAIN_MOTOR);
 
 added line delay
 You can fix the example file

corneliusmunz, 2021-01-06:
If you want to use the CONTROL_PLUS_HUB you have also to use the Port definitions in the enum `ControlPlusHubPort` So the line of the attachDevice should look as follows:
```c++
myEmulatedHub.attachDevice ((byte) ControlPlusHubPort :: D, DeviceType :: TRAIN_MOTOR);
```

corneliusmunz, 2021-01-06:
@smotek Regarding the delays: In the current version the delays are already adjusted. Which version of the library do you use?

smotek, 2021-01-07:
Downloaded library 9 days ago.
The problem is not in the library, but in the HubEmulation example.
[image]

function
[image]

After editing to ControlPlusHubPort, ports A - D work

Another question, what it means in a line
pf.single_pwm(PowerFunctionsPort::RED, pf.speedToPwm(value)); 

PowerFunctionsPort::RED ?

corneliusmunz, 2021-01-07:
The `PowerFunctionsPort::RED` is addressing the "RED" IR channel on the power functions IR receiver. There is a RED and a BLUE channel on the receiver available. The example receives signals from the PoweredUp App and forwards it per IR LED and the PowerFunctions methods to the IR receiver.

smotek, 2021-01-07:
Now I understand
I forgot that the example sends an IR to an old cube.

Chris-Rodrigues, 2021-08-15:
Hello! Congratulations to all persons who are working on this project. Very nice.
Smotek helped me in other forum with this code so I found this forum now. Thanks Smotek.

I succefully ran Lpf2HubEmulation and now I'm trying to emulate two or more hubs in just one ESP32. 
I still have a lot of free pins from ESP32 and using one app I want to emulate more hubs to control more things at same screen.

On app, first appears HUB1. After some time, name changes to HUB2 but I can't have both at same time

I don't know if am I doing or saying a dumb thing, but here it is:

Lpf2HubEmulation myEmulatedHub("HUB1", HubType::CONTROL_PLUS_HUB);
Lpf2HubEmulation myEmulatedHub2("HUB2", HubType::CONTROL_PLUS_HUB);

void writeValueCallback(byte port, byte value) {
...
}
void writeValueCallback2(byte port, byte value) {
...
}

void setup(){
  Serial.begin(115200);
  myEmulatedHub.setWritePortCallback(&writeValueCallback);
  myEmulatedHub.start();
  
  myEmulatedHub2.setWritePortCallback(&writeValueCallback2);
  myEmulatedHub2.start();

...
}

void loop() {

  if (myEmulatedHub.isConnected && !myEmulatedHub.isPortInitialized)  {
    delay(1000);
    myEmulatedHub.isPortInitialized = true;
    delay(1000);
    myEmulatedHub.attachDevice((byte)ControlPlusHubPort::A, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub.attachDevice((byte)ControlPlusHubPort::B, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub.attachDevice((byte)ControlPlusHubPort::C, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub.attachDevice((byte)ControlPlusHubPort::D, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub.attachDevice((byte)ControlPlusHubPort::LED, DeviceType::HUB_LED);
    delay(1000);
  }

  if (myEmulatedHub2.isConnected && !myEmulatedHub2.isPortInitialized) {
    delay(1000);
    myEmulatedHub2.isPortInitialized = true;
    delay(1000);
    myEmulatedHub2.attachDevice((byte)ControlPlusHubPort::A, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub2.attachDevice((byte)ControlPlusHubPort::B, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub2.attachDevice((byte)ControlPlusHubPort::C, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub2.attachDevice((byte)ControlPlusHubPort::D, DeviceType::TRAIN_MOTOR);
    delay(1000);
    myEmulatedHub2.attachDevice((byte)ControlPlusHubPort::LED, DeviceType::HUB_LED);
    delay(1000);
  }
}

paki111, 2021-10-06:
@Chris-Rodrigues interesting idea, but this may be not posible on one MAC address.  I think the solution for having more ports (6 vs 4) is to implement support for Spike Prime Hub. What do you think @corneliusmunz ? 
BTW @corneliusmunz thank's you for great library!

corneliusmunz, 2021-10-06:
@Chris-Rodrigues I have never tried this and I am not sure if the NimBle library does support multiple connections. If you use the library to connect to other hubs it ist possible to have up to 9 connections at the same time. But I have never tried it the other way round. Maybe I will find some time in the next days to try it out