corneliusmunz/legoino #53: Using legoino together with esp32-ps3 not possible?

fredlcore, 2021-03-08:
I would like to use my old PS3 controller together with legoino and found this library here: https://github.com/jvpernis/esp32-ps3
However, if I use their "Ps3Demo" from the example folder and simply add
```
#include "Lpf2Hub.h"
// create a hub instance
Lpf2Hub myHub;
```
at the top of the file and 
`myHub.init()` 
in the setup() function, the script either crashes with this message if I add the init() function after the PS3-related calls:
```
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x40091b64
file: "/Users/frederik/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLEDevice.cpp" line 559
func: static void NimBLEDevice::init(const string&)
expression: esp_bt_controller_init(&bt_cfg)
```
Or throws this error (if myHub.init() goes before the PS3 attach functions) without crashing, but also without connecting to the PS3 controller anymore:
```
CRIT NimBLEDevice: "Resetting state; reason=19, "
```

Is it technically not possible to use legoino's BLE connection together with Esp32-Ps3's "classic" serial connection to the PS3 controller? Or is this something that could be resolved in code?
I'm referencing @jvpernis here as well just in case.

corneliusmunz, 2021-03-09:
Hi @fredlcore Again a cool idea 😄 I have had a short look into the https://github.com/jvpernis/esp32-ps3 library and i think it is a problem of the BLE stack. Legoino uses the following BLE Library https://github.com/h2zero/NimBLE-Arduino and the ps3 library uses the Bludroid stack with a own implementation of the GATT protocol. I think these two libaries does not fit together and wants to access the same BLE resources of the espressif ble stack. Unfortunately i think this will be a huge effort on one of the libs to port the BLE stack implementation. At the end it should work that the ps3 libary uses the NimBLE-Arduino lib and write and adapter for the used functions/callbacks. @jvpernis Do you think that this would be possible? The NimBLE-Arduino stack gains attention and replaces step by step the no longer maintained https://github.com/nkolban/ESP32_BLE_Arduino

fredlcore, 2021-03-09:
Thanks for the background information, @corneliusmunz! It would be really cool to get this working on one and the same ESP32 because now I have to make two ESP32s talk via serial to each other, one is handling the PS3, the other one the Legoino part. It works quite well, but it's of course a waste of resources and also (electrical) power. The great thing about this solution is that I can just put the ESP somewhere to a wall outlet and the kids can use the PS3 control without any cables and it brings the two analog sticks which make fine-grained control of the triain possible :)...

corneliusmunz, 2021-03-09:
@fredlcore Another possibility (which i have used for my kids) is the usage of the Lego PoweredUp remote control which is shipped with the Lego trains (https://www.lego.com/de-de/product/remote-control-88010) . This is quite cheap and you can use it with legoino to control different devices. You can find a example here: https://github.com/corneliusmunz/legoino/blob/master/examples/PoweredUpRemote/PoweredUpRemote.ino  and here: https://youtu.be/UtHMKe2Insw

fredlcore, 2021-03-09:
That's also a good option, but it doesn't seem to have an analog control, does it? That is one of the really cool features of the PS3 controller because it makes the train behave more like a real train where it stops when you let go of the stick. 
But I think I'll get one of those as well because they fit in more nicely :)...