corneliusmunz/legoino #57: Best approach to test if connection is still active?

fredlcore, 2021-04-05:
I would like to detect whether and established connection is still active or whether the device (here: Duplo train) has been turned off. So far, if I turn off the device, `myHub.isConnected()` continues to evaluate as true.

Is there some kind of variable or characteristic I can read that is continuously updated and from which I can detect whether the connection is still active? Or can I send a command to the device that will generate a response?

And once I have figured out that the device has been turned off, what actions do I need to take to get my code back into listening mode? Do I have to do a `myhub.init()` again? Or can I somehow "close" the connection and then myHub will be in scanning mode again?

Thanks!

fredlcore, 2021-04-05:
Hm, after a bit more testing, it seems that isConnected() seems to be switching relatively quick to `false`once I have turned of the connected device, so that should do what I want/need. Nevertheless, if there are any other things to make this process more reliable, I'm happy to hear!

closed 2021-04-05 by fredlcore

corneliusmunz, 2021-04-06:
Hi @fredlcore !  Thats so far exact the procedure i used.

1. check if `myHub.isConnected() == false`
2. call myHub.init()

I have used this in the `TwoTrainHubs.ino` example sketch in the main loop: https://github.com/corneliusmunz/legoino/blob/434b74fe6717640a9d054ef7533d68c764e33a5d/examples/TwoTrainHubs/TwoTrainHubs.ino#L145

Btw, if you like the library i would be happy if you give me a :star: 😉 

Best regards,
Cornelius

fredlcore, 2021-04-07:
Starred it is :)!