corneliusmunz/legoino #58: Add a status variable/function to determine whether a non-blocking scan is still active

fredlcore, 2021-04-05:
The default duration for a scan is 10 seconds. If you start the ESP32, it will scan for that duration for any kind of device, but not after these 10 seconds. So if you turn on your device after that period, it won't connect unless you restart the ESP32.
If one could test whether the non-blocking scan is over, one could restart the scan by calling myHub.init() again afterwards.

I tried to add the variable _isScanning and set it to true in the start() function and set it again to false in the scanEndedCallback() function, but somehow these are in different namespaces and I can't pass a pointer to _isScanning, so it didn't work out with my limited C++ knowledge. Do you have any idea wheter this could be achieved properly? Thanks!

corneliusmunz, 2021-04-06:
You can increase the scan time by using the `void init(uint32_t scanDuration);` overloaded init function with `scanDuration` in seconds. If you want to use the callback function if a scan has ended, you have to Subclass the callback function and give the hub instance as a parameter to the subclassed function to "remember" the instance which has initiated the scan process. The same was done for the `onConnect()` and `onDisconnect()` callbacks: 
https://github.com/corneliusmunz/legoino/blob/434b74fe6717640a9d054ef7533d68c764e33a5d/src/Lpf2Hub.cpp#L31

The callbacks are used in the following line of code with the `this` keyword to remember the current hub instance and to have access to the member variables (like you tried it with `_isScanning`) 
https://github.com/corneliusmunz/legoino/blob/434b74fe6717640a9d054ef7533d68c764e33a5d/src/Lpf2Hub.cpp#L1160

fredlcore, 2021-04-06:
Great, thanks, my object oriented knowledge is not too extensive, but I think I know what you say and it's exactly what I need. Can I somehow subclass the callback function in my own code somehow so that it is used instead of the one provided in your library? Or do I have to modify your library source code (which I would like to avoid)?

corneliusmunz, 2021-04-06:
You can do this in your own code but you have to adapt the library to use your implementation of the callback function in the following line:
https://github.com/corneliusmunz/legoino/blob/434b74fe6717640a9d054ef7533d68c764e33a5d/src/Lpf2Hub.cpp#L823

So you can try it locally but i will think of an additional feature to use the scan ended callback and expose a `_isScanning` property in the library. Unfortunately the time working on the library is a little bit rare.

fredlcore, 2021-04-07:
Of course I don't want to take up any more of your time if I can do it myself, @corneliusmunz - I think I know how to expose _isScanning, my only stumbling block is that I don't really understand in which namespace "scanEndedCallback" resides. It's apparently not in the Lpf2Hub namespace, because if I declare _isScanning as a class variable, I can use it within other Lpf2Hub methods, but not inside "scanEndedCallback". If I move "scanEndedCallback" inside the Lpf2Hub section and prefix it with "Lpf2Hub::", I get an error (among others) that it needs to be declared as static. That is where my knowledge of object-oriented programming reaches its limits ;). In case this is easy to fix, I can happily provide the remaining code for exposing _isScanning and send you a PR so you don't have to worry about it.

fredlcore, 2021-04-07:
Ok, I managed to solve the problem myself :) - not via the callback function (that one is still beyond my comprehension), but I figured out that BLEScan already provides a isScanning method that we just need to make public. My PR #60 makes these little changes and should hopefully work without any problems. At least it does here :).

fredlcore, 2021-04-07:
@corneliusmunz: Some of the CI tests fail, but according to the logs, that does not seem to have to do with my code, I guess...

corneliusmunz, 2022-01-08:
@fredlcore fixed with your PR #60

closed 2022-01-08 by corneliusmunz