corneliusmunz/legoino #64: Bug in LPF2Hub.cpp

ThorstenBenter, 2021-12-15:
Hi Cornelius,
I tried to get in touch with you on the Eurobricks Forum (https://www.eurobricks.com/forum/index.php?/forums/topic/180905-new-version-100-of-legoino-arduino-library/&do=findComment&comment=3478705) - but I am sure you have simply so many other things to do. 
Just to make sure: Legoino is my absolutely favorite software, when it comes to PF and PF2 control. It does not get >any< better. I have no clue how you accomplished this besides job and life. Thank you again so much for providing this to the community. With your help and efforts , I was able to make a dream come true (https://www.eurobricks.com/forum/index.php?/forums/topic/188584-mulpi-a-multiple-lego-remote-protocol-interface/).

I believe there are two minor bugs in LPF2Hub.cpp, please see the first Eurobricks link above for details.
In short:

[I believe it is favorable - even in the function names - to distinguish "power" from "speed". TLG does that in their totally outdated LWP3.0 protocol document on GitHub as well. That is just naming, but will break code, but there are very simple workarounds]

(1) For the setTachoMotorSpeed function, you are using 0x01 as motor sub-command of the LWP3.0 0x81 Output Command. I believe this needs to be 0x07 [StartSpeed (Speed, MaxPower, UseProfile)] to work correctly.
(2) The StartSpeed sub-command does not have any BrakeStyle option - as it is controlling the speed. All other regulated (speed) commands do have that option and work beautifully well!
(3) Once the setTachoMotorSpeed function is changed to work with sub-command 0x07, the stopTachoMotor function does not work anymore, as there the speed setting 0 does cause the function to behave weird (it ramps the power to 100%). The LWP3.0 protocol says that when you want to set the speed of a tacho motor to 0, you have to use StartPower(port, 0(brake)/127(float)) to stop the motor.

For the setBasicMotorSpeed, you are using sub-command 0x51. I believe, it would be more consistent to use 0x01 (StartMotorPower). But: That is just a matter of taste, but not functionality. 0x51 works perfectly well.

That's all - just very minor things. I have tested the changed cpp file extensively and - it seems to work.

All the best,
Thorsten

corneliusmunz, 2021-12-16:
Hi Thorsten (@ThorstenBenter)! Many thanks for reaching out via GitHub 👍  I will check about the different commands for the Tacho motors. Very good finding! On the time i have implemented this i was not really able to test it with the appropriate hardware. 
Can you share your adaptions to your code in your cpp files? I have some other open issues and hopefully during christmas holidays i will find some time for maintaining the library and releasing a new version.

All the best,
Cornelius

ThorstenBenter, 2021-12-16:
Hi Cornelius (@corneliusmunz ),
thank you for your swift reply! That would be very nice - however, holiday are holidays for a reason ;)
I am absolutely happy to share what I have changed - but there are now stupid comments in the files ... as said, I feel comfortable with assembler for (>very< small) RISC microcontrollers or - well - BASIC :D What would be a good way to get the files to you? I am not familiar with the GitHub environment other than downloading from this site ... oh wait: A line below it says "Attach files by ..." - would that be OK?

ThorstenBenter, 2021-12-16:
Hi Cornelius (@corneliusmunz)
I have assembled all the stuff that may or may not be of help with regard to 
a) the motor commands and
b) color changes
c) the isScanning function as suggested by @fredlcore

(1) Changes.txt is just my personal log to keep track of things I did to original files
(2) Keywords.txt is just having the isScanning keyword added
(3) LPF2Hub.cpp/h are the files I am currently using for the follwing test program:
(4) LegoinoTrainHubTest.ino
[LegoinoStuff.zip](https://github.com/corneliusmunz/legoino/files/7729910/LegoinoStuff.zip)

I guess this is all I have.

All the best,
Thorsten

corneliusmunz, 2022-01-09:
Hi  Thorsten! (@ThorstenBenter),
i have reviewed, evaluated and tested your changes and created a feature branch with my changes. I have almost copied all changes from your files but have made some small adaptions. For compatibility reasons I have kept the "stopTachoMotor" and "stopBasicMotor" stuff and also the setBasicMotorSpeed (even if this makes no sense as you have perfectly described it in your files). Additionally I have added the "stopMotor" command which should be in the future the common command for all motors. Also the setMotorPower function is added because this ist the "correct" naming. 

Regarding the colors I will change this in a "breaking" release and will introduce a struct enum where you can define the color like in other struct enums "Lpf2::RED" etc. 

The PR from @fredlcore is already merged into the master branch and I have added the isScanning keyword afterwards as you have suggested it. 

Here you can find the feature branch with the changes I have made related to your issue: https://github.com/corneliusmunz/legoino/tree/feature/issue_64_wrong_motor_command

It would be great if you could do a review on the changes and could test if this will solve your issues. I would highly appreciate this. 

All the best and a littlebit late a happy new year 🥳 
Cornelius

fredlcore, 2022-01-09:
@ThorstenBenter: cool project! And even cooler that you used your Specci :) - I'm an 8-bit enthusiast as well (coming from Atari), and maybe just as an idea if you want to open your project for people without the interface 1: it's fairly easy to communicate with an Arduino/ESP from an 8-Bit computer using any kind of joystick port or user interface. I've made a little project called AtariDuino for that, maybe it's helpful if you want to think in that direction:
https://github.com/fredlcore/AtariDuino

Is the ZX code in BASIC or Assembly? The former would be easy to port to the Atari and other 8-Bit systems, Z80 code not so much ;)...

ThorstenBenter, 2022-01-09:
Hi Cornelius! (@corneliusmunz) 
Happy new year to you and your family as well! 
Thank you very much again for looking into this - I really do highly appreciate that. Legoino is super stable (with regard to both PF and PUp) and I am happily coding stuff into my ESP32 using "BASIC C++" style :D To be honest, Legoino allows me not only to continue with my now 21 years long project, but it means anyone who wants, can do sophisticated controlling/monitoring of current PUp devices for less than 10€ = one ESP32 board. For me, this is simply phantastic!
All the best, Thorsten

ThorstenBenter, 2022-01-09:
@fredlcore Thank you very much!
Oh that sounds very good - I'll have a look! Thank you for that link!
The ZX code is in BASIC. My Z80 assembler capabilities have become a bit rusty over the past 30 years, but I am planning of implementing a couple of small assembler code bits to e.g. speed up critical timing or getting direct access to the 8k screen memory. Back in the olden days (and more recently it all came back :D) I was much more interested in fudging with the hardware of the Speccy. Here is my personal "log" of what I've done so far: https://uni-wuppertal.sciebo.de/s/dcmzcdwNoQv63h4.
Thanks again, and have a very nice day!
All the best
Thorsten