corneliusmunz/legoino #65: LegoinoCommon.h missing #include "string"

tsochatzki, 2022-01-11:
Hi Cornelius,
while trying the trainhub.ino i get the following issue:
In file included from C:\Users\tsoch\sketchbook\libraries\Legoino\src\LegoinoCommon.cpp:11:
C:\Users\tsoch\sketchbook\libraries\Legoino\src\LegoinoCommon.h:30:15: error: 'string' in namespace 'std' does not name a type
   static std::string ColorStringFromColor(Color color);
               ^~~~~~
C:\Users\tsoch\sketchbook\libraries\Legoino\src\LegoinoCommon.h:30:10: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
C:\Users\tsoch\sketchbook\libraries\Legoino\src\LegoinoCommon.h:16:1:
+#include <string>

After updating the LegoinoCommon.h with the    #include "string"       it works without any issue.

Regards, Thomas

corneliusmunz, 2022-01-11:
Hi Thomas (@tsochatzki) Many thanks for that finding! I will check this and add the missing include in the next release

corneliusmunz, 2022-01-11:
@tsochatzki  which version of the Arduino Esp32 have you installed? Do you use platform.io as environment or the Arduino IDE itself
[image]
?

tsochatzki, 2022-01-12:
Hi Cornelius,
iam using the Arduino IDE and the following ESP32 Version: 2.0.2

M9lab, 2026-02-10:
I solved with:

LegoinoCommon.h
++ #include <string> (row 16)

LegoinoCommon.cpp
Corrected the return types of the functions ReadUInt8, ReadInt8, ReadUInt16LE, ReadInt16LE, ReadUInt32LE, ReadInt32LE.

Changed them from uint8_t, int8_t, etc. to unsigned char, signed char, etc. to match the header.

Removed the default values = 0 from the implementations (they must appear only in the header).