corneliusmunz/legoino #70: Problems with getting started following the Quickstart tutorial

Nielceo, 2022-08-04:
Apologies for probably a basic question. My son and me try to follow your Quickstart tutorial using an M5 ATOM Matrix ESP32 board. When we try to compile the TrainHub program we get the following error messages. Any idea what we do wrong? 

Arduino: 1.8.19 (Mac OS X), Board: "Arduino Uno"

WARNING: library Legoino claims to run on esp32 architecture(s) and may be incompatible with your current board which runs on avr architecture(s).
TrainHub:13:1: error: 'Lpf2Hub' does not name a type
 Lpf2Hub myTrainHub;
 ^~~~~~~
TrainHub:14:19: error: 'PoweredUpHubPort' has not been declared
 byte port = (byte)PoweredUpHubPort::A;
                   ^~~~~~~~~~~~~~~~
/Users/cornelis/Documents/Arduino/libraries/Legoino/examples/TrainHub/TrainHub.ino: In function 'void loop()':
TrainHub:24:8: error: 'myTrainHub' was not declared in this scope
   if (!myTrainHub.isConnected() && !myTrainHub.isConnecting())
        ^~~~~~~~~~
TrainHub:31:7: error: 'myTrainHub' was not declared in this scope
   if (myTrainHub.isConnecting()) {
       ^~~~~~~~~~
TrainHub:45:7: error: 'myTrainHub' was not declared in this scope
   if (myTrainHub.isConnected()) {
       ^~~~~~~~~~
TrainHub:50:28: error: 'GREEN' was not declared in this scope
     myTrainHub.setLedColor(GREEN);
                            ^~~~~
/Users/cornelis/Documents/Arduino/libraries/Legoino/examples/TrainHub/TrainHub.ino:50:28: note: suggested alternative: 'SREG_N'
     myTrainHub.setLedColor(GREEN);
                            ^~~~~
                            SREG_N
TrainHub:52:28: error: 'RED' was not declared in this scope
     myTrainHub.setLedColor(RED);
                            ^~~
exit status 1
'Lpf2Hub' does not name a type

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

corneliusmunz, 2022-08-05:
Hi @Nielceo! No worries you can just ask :-) This error message means, that you have selected the wrong board in your Arduino IDE (Integrated Development Environment) Here you can find the configuration setup for the ATOM Matrix Esp32 controller: https://docs.m5stack.com/en/quick_start/atom/arduino I think you can start directly in with the section "M5Stack's board management" Hope this will help you. If you have any further questions, just write it here back in GitHub

Nielceo, 2022-08-06:
Great! It works now. Thanks!

closed 2022-08-07 by corneliusmunz