corneliusmunz/legoino #82: error example lego trainhub

Timo-VB, 2024-12-01:
It doesn't work i am getting errors if i want to upload the example.

uwemon, 2024-12-01:
Hello, you are using NimBLE-Arduino Library 2.0.0. Go back to version 1.4.3. and try again.

Best regards

Uwe

Timo-VB, 2024-12-01:
It still doesn't work. He is giving the same error.

corneliusmunz, 2024-12-02:
Hi @Timo-VB There is indeed an error when using the latest Nimble Library. If you have problems even after downgrading to NimBLE-Arduino 1.4.3, just give me some more details about the error.

Which development library have you used (Arduino IDE or PlatformIO)?
Which board are you using 
Can you send me the detailed error?

Best regards,
Cornelius

Maybe I have time during the Christmas holidays to prepare the library for the NimBLE-Arduino update

corneliusmunz, 2024-12-02:
Hi @Timo-VB ! I have tried to reproduce the issue on my system and I think I have found the error. The error ist related to the ESP32 library which has the current version 3.x.x  In the time where I have developed the library, the version 2.x.x was available. In the version change a breaking change was introduced about string handling: https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html 

So as long as I have no new version, you can

 try to compile the example sketches with the latest 2.x.x version of the esp32 framework (2.0.17) With this version I can compile without any errors and with the 3.x.x versions I got the same error as you. 

Can you please give this a try? You can change the ESP32 framework version as follows:

[image]

Timo-VB, 2024-12-03:
It works! Thank you very very much!

Best regards,
Timo-VB

closed 2024-12-03 by Timo-VB