corneliusmunz/legoino #83: Many problems with Lego Duplo Train using ESP32

protusi, 2024-12-01:
Hi!

Can`t compile (use Arduino IDE) project for  Lego Duplo Train using ESP32. How can I fix it?

> In file included from /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.cpp:11:
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.h:29:15: error: 'string' in namespace 'std' does not name a type
>    static std::string ColorStringFromColor(Color color);
>                ^~~~~~
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.h:29:10: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.h:16:1:
> +#include <string>
>  
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.h:29:10:
>    static std::string ColorStringFromColor(Color color);
>           ^~~
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.h:30:15: error: 'string' in namespace 'std' does not name a type
>    static std::string ColorStringFromColor(int color);
>                ^~~~~~
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.h:30:10: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
>    static std::string ColorStringFromColor(int color);
>           ^~~
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.cpp:39:6: error: 'string' in namespace 'std' does not name a type
>  std::string LegoinoCommon::ColorStringFromColor(Color color)
>       ^~~~~~
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.cpp:39:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.cpp:12:1:
> +#include <string>
>  
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.cpp:39:1:
>  std::string LegoinoCommon::ColorStringFromColor(Color color)
>  ^~~
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.cpp:48:6: error: 'string' in namespace 'std' does not name a type
>  std::string LegoinoCommon::ColorStringFromColor(int color)
>       ^~~~~~
> /Users/protusi/Documents/Arduino/libraries/Legoino/src/LegoinoCommon.cpp:48:1: note: 'std::string' is defined in header '<string>'; did you forget to '#include <string>'?
>  std::string LegoinoCommon::ColorStringFromColor(int color)
>  ^~~
> 
> exit status 1
> 
> Compilation error: exit status 1

corneliusmunz, 2024-12-02:
Hi @protusi  I will try to have a look on this in the next days. Can you provide me some more details about the error?

Which development environment have you used (PlatformIO, Adruino IDE, ... )?
Which type of board have you used
Which NimBLE-Arduino library version is installed or referenced?

Best regards,
Cornelius

corneliusmunz, 2024-12-02:
Hi @protusi  can you try the following thing: https://github.com/corneliusmunz/legoino/issues/82#issuecomment-2512934696 I think it is the same issue as you have with the same error logs. It is related to a breaking change in the esp32 framework library from 2.x.x to 3.x.x

protusi, 2024-12-03:
Hi guys. Thanx for your answers. Problem was in version of NimBLE. With 1.4.3 - everything is ok. Thanks!

corneliusmunz, 2024-12-03:
@uwemon Many thanks for your support! That ist highly appreciated! @protusi Good to hear, that it works now. I will try to update legoino to specify the NimBLE dependency in a more clear way

protusi, 2024-12-03:
> @uwemon Many thanks for your support! That ist highly appreciated! @protusi Good to hear, that it works now. I will try to update legoino to specify the NimBLE dependency in a more clear way

thanks for legoino 😀 great project

closed 2024-12-03 by corneliusmunz