corneliusmunz/legoino #84: button doesn't work

Timo-VB, 2024-12-05:
I am using lpf2hub en that works but if I want to use it with a button nothing works anymore, I can't connect to the train and the button doesn't work.

corneliusmunz, 2024-12-05:
Hi @Timo-VB ! Can you show me your code? What do you mean with button? A button connected to the esp32 board?

Timo-VB, 2024-12-06:
This is my code:
#include "Lpf2Hub.h"

// create a hub instance
Lpf2Hub myTrainHub;
byte port = (byte)PoweredUpHubPort::A;

const byte drukknoprijden = 16;
const byte drukknopstop = 4;
const byte rodeLed = 23;
const byte groeneLed = 22;

void setup() {
  Serial.begin(115200);
  pinMode(drukknoprijden, INPUT);
  pinMode(drukknopstop, INPUT);
  pinMode(rodeLed,OUTPUT);
  pinMode(groeneLed,OUTPUT);

  digitalWrite(groeneLed, LOW);
  digitalWrite(rodeLed, HIGH);
} 

// main loop
void loop() {

  if (!myTrainHub.isConnected() && !myTrainHub.isConnecting()) 
  {
    myTrainHub.init(); // initalize the PoweredUpHub instance
    //myTrainHub.init("90:84:2b:03:19:7f"); //example of initializing an hub with a specific address
  }

  // connect flow. Search for BLE services and try to connect if the uuid of the hub is found
  if (myTrainHub.isConnecting()) {
    myTrainHub.connectHub();
    if (myTrainHub.isConnected()) {
      Serial.println("Connected to HUB");
      Serial.print("Hub address: ");
      Serial.println(myTrainHub.getHubAddress().toString().c_str());
      Serial.print("Hub name: ");
      Serial.println(myTrainHub.getHubName().c_str());
    } else {
      Serial.println("Failed to connect to HUB");
    }
  }

  if (myTrainHub.isConnected()) {
    char hubName[] = "myTrainHub";
    myTrainHub.setHubName(hubName);
  }

  if(digitalRead(drukknoprijden) == HIGH){
    myTrainHub.setBasicMotorSpeed(port, 35);
    digitalWrite(groeneLed, HIGH);
    digitalWrite(rodeLed, LOW);
    Serial.print("gedrukt");
  }

  if(digitalRead(drukknopstop) == HIGH){
    myTrainHub.stopBasicMotor(port);
    digitalWrite(groeneLed, LOW);
    digitalWrite(rodeLed, HIGH);
  }
  // if connected, you can set the name of the hub, the led color and shut it down
  /*if (myTrainHub.isConnected()) {
    char hubName[] = "myTrainHub";
    myTrainHub.setHubName(hubName);
  
    myTrainHub.setLedColor(GREEN);
    delay(1000);
    myTrainHub.setLedColor(RED);
    delay(1000);
    myTrainHub.setBasicMotorSpeed(port, 35);
    delay(1000);
    myTrainHub.stopBasicMotor(port);
    delay(1000);
    myTrainHub.setBasicMotorSpeed(port, -35);
    delay(1000);
    myTrainHub.stopBasicMotor(port);
    delay(1000);

  } else {
    Serial.println("Train hub is disconnected");
  }
  */
  
} // End of loop

I mean a button connected to my esp32.

corneliusmunz, 2024-12-06:
What type of board do you use?

Timo-VB, 2024-12-06:
I am using an esp32 dev module.

corneliusmunz, 2024-12-09:
Hi @Timo-VB  
I think I have found the issue. Your code with the button is running inside the loop without checking  the condition, if the hub is connected or not. You should set the commands for the hub only if it is connected. So you have to wrap the button code inside an `if (myTrainHub.isConnected()) {...}` code block.

Additionally the button needs maybe some kind of debouncing. 

I have tried the following code on an M5 Atom - Matrix ESP32 board. There the button is connected to the Port 39. In your case you have to change it to Pin 4, 16, 22, 23 dependent what you want. So my suggestion is, that you take the following code. Change the pin to one of your buttons and try to run it without any other changes. Then you should be able to start and stop the train motor with your button. If this works, you can change further things.

Example Code:

```c++
/**
 * A Legoino example to control a train which has a motor connected
 * to the Port A of the Hub
 * 
 * (c) Copyright 2020 - Cornelius Munz
 * Released under MIT License
 * 
 */

#include "Lpf2Hub.h"

// create a hub instance
Lpf2Hub myTrainHub;
byte port = (byte)PoweredUpHubPort::A;
const byte buttonPin = 39;
bool isMotorRunning = false;

void setup() {
  Serial.begin(115200);
  pinMode(buttonPin, INPUT_PULLUP);
}

// main loop
void loop() {

  if (!myTrainHub.isConnected() && !myTrainHub.isConnecting()) {
    myTrainHub.init();  // initalize the PoweredUpHub instance
    //myTrainHub.init("90:84:2b:03:19:7f"); //example of initializing an hub with a specific address
  }

  // connect flow. Search for BLE services and try to connect if the uuid of the hub is found
  if (myTrainHub.isConnecting()) {
    myTrainHub.connectHub();
    if (myTrainHub.isConnected()) {
      Serial.println("Connected to HUB");
      Serial.print("Hub address: ");
      Serial.println(myTrainHub.getHubAddress().toString().c_str());
      Serial.print("Hub name: ");
      Serial.println(myTrainHub.getHubName().c_str());
    } else {
      Serial.println("Failed to connect to HUB");
    }
  }

  // if connected, you can set the name of the hub, the led color and shut it down
  if (myTrainHub.isConnected()) {

    if (digitalRead(buttonPin) == LOW) {
      Serial.println("Button pressed...");
      if (isMotorRunning == true) {
        Serial.println("Stop motor...");
        myTrainHub.stopBasicMotor(port);
        isMotorRunning = false;
        delay(200); // debounce
      } else {
        Serial.println("Start motor...");
        myTrainHub.setBasicMotorSpeed(port, 35);
        isMotorRunning = true;
        delay(200); // debounce
      }
    }

  } else {
    Serial.println("Train hub is disconnected");
  }

}  // End of loop
```

Timo-VB, 2024-12-17:
It still doesn't work. There is no error but if i clik it don't do anything. If i connect the hub he start riding but doesn't stop if i click. If i click multiple times it also doen't do anything. Can the problem be my board or not?

corneliusmunz, 2024-12-19:
Hi @Timo-VB ! Good Question. Could be your Board or your wiring. What exact type of board do you use? Sometimes some pins are used for other purposes. Which gpio pin do you use?

Timo-VB, 2024-12-19:
Hi

On my board there stand it is an esp32 wroom 32d but in arduino ide i can't find that so i select esp32 dev module.

[image]

corneliusmunz, 2024-12-19:
Ok, I think the board is ok and also the selection of the board type in the Arduino IDE. I see, that you have do the wiring in that way, that you have a pulldown resistor to your input port. That means, if you press the button, you have a high signal. This was an inverse logic I have used in my wiring. Can you please try the following code (Changed input port configuration that it fits to your wiring)

```c++
/**
 * A Legoino example to control a train which has a motor connected
 * to the Port A of the Hub
 * 
 * (c) Copyright 2020 - Cornelius Munz
 * Released under MIT License
 * 
 */

#include "Lpf2Hub.h"

// create a hub instance
Lpf2Hub myTrainHub;
byte port = (byte)PoweredUpHubPort::A;
const byte buttonPin = 23;
bool isMotorRunning = false;

void setup() {
  Serial.begin(115200);
  pinMode(buttonPin, INPUT);
}

// main loop
void loop() {

  if (!myTrainHub.isConnected() && !myTrainHub.isConnecting()) {
    myTrainHub.init();  // initalize the PoweredUpHub instance
    //myTrainHub.init("90:84:2b:03:19:7f"); //example of initializing an hub with a specific address
  }

  // connect flow. Search for BLE services and try to connect if the uuid of the hub is found
  if (myTrainHub.isConnecting()) {
    myTrainHub.connectHub();
    if (myTrainHub.isConnected()) {
      Serial.println("Connected to HUB");
      Serial.print("Hub address: ");
      Serial.println(myTrainHub.getHubAddress().toString().c_str());
      Serial.print("Hub name: ");
      Serial.println(myTrainHub.getHubName().c_str());
    } else {
      Serial.println("Failed to connect to HUB");
    }
  }

  // if connected, you can set the name of the hub, the led color and shut it down
  if (myTrainHub.isConnected()) {

    if (digitalRead(buttonPin) == HIGH) {
      Serial.println("Button pressed...");
      if (isMotorRunning == true) {
        Serial.println("Stop motor...");
        myTrainHub.stopBasicMotor(port);
        isMotorRunning = false;
        delay(200); // debounce
      } else {
        Serial.println("Start motor...");
        myTrainHub.setBasicMotorSpeed(port, 35);
        isMotorRunning = true;
        delay(200); // debounce
      }
    }

  } else {
    Serial.println("Train hub is disconnected");
  }

}  // End of loop
```

Timo-VB, 2024-12-19:
Hi 

I have already tried that.

corneliusmunz, 2024-12-19:
Can you get the button work without any legoino stuff? Maybe just a Serial.println("button pressed") statement. If not, something with your board is wrong