corneliusmunz/legoino #87: Compability with NimBLE-Arduino 2.x

SFrijters, 2025-03-13:
Recently, NimBLE-Arduino has been largely overhauled and legoino is not compatible, but it would be nice if it would be.

See https://github.com/h2zero/NimBLE-Arduino/releases/tag/2.1.0 .

carlalldis, 2025-03-23:
I second this request please. I have tried to do it myself by replacing the deprecated functions, but it seems there is more to it. A lot of it might be the new usage of std::vector<uint8_t> instead of std::string for data methods.

In the meantime i would like to use an older version of NimBLE + Legoino but I am strugglgin to find a working combination (ESP32 immediately crashing). Does anyone have a good version combo?

SFrijters, 2025-03-23:
You may want to look at https://github.com/Rbel12b/legoino or my own fork https://github.com/SFrijters/legoino . I haven't had time to test it on hardware yet, but the compilation test suite passes.

carlalldis, 2025-03-24:
Thanks @SFrijters, that seems to work once it's connected. I'm just having intermittent success in getting the initial connection to my ConnectPlus hub - requires a bunch of restarts before the connection is successful, but from then it's rock solid.

I'll try to enable debug logging when i have chance.

SFrijters, 2025-03-26:
Great, let me know if you find anything.
I'll also try it on my train hubs when I have time, and if it all seems to work well enough I'll open a PR here so people can keep using the original library.

SFrijters, 2025-04-10:
I've tested the TrainHub example on real hardware, both the example code and the more elaborate code I'm actually using and that all seems to work.

As said above, I'll be happy to make a PR for the compatibility changes, but since some other PRs have been lingering for a long time I'd like to hear from @corneliusmunz first.

Stuckalf, 2025-04-29:
Hi, I tried the fork from SFrijters but couldn't get any bluetooth connection to the Lego Remote 88010. What a shame. Last year in the first tests all work out well and I had a steady connection between Wemos D1 Mini ESP32 and my Lego Remote. 

I had to downgrade the Arduino ESP libraries to get it compiled. I remove the original Legoino fork from Cornelius and installed the fork from SFrijters, but I get no connection at all. Anything I have to do additionally to get the connection up and running?

carlalldis, 2025-04-29:
Sounds like the same issue I was having. I did try to grab some debug logs, but the library still has some old non-working references to the NimBLE 1.x API when debug level 5 is enabled and I never got chance to correct these.

Stuckalf, 2025-04-29:
I'm sorry, it was my own fault. 
Last year I tested Legoino with an ESP32 C3 Super Mini.
This time I tried a Wemos D1 Mini. It compiled, but the BLE connection is unstable, to say the least. Often the ESP32 doesn't even conenct at all to the Remote Controll.
And after the script is running on the ESP, the Arduino IDE built in serial monitor freezes with a strange message like:
///////////////////////////////////////////////////////////////////////////////// and some other characters.
It is strange that the script is running fine in the background - I can see the ESP's reaction if a Remote Control button is pressed.
But the serial monitor doesn't show the awaited messages.

I compiled a new test for an ESP32 C3 Super Mini and all problems are gone (at least as far as I could check it).

It would be nice to get it running on the Wemos D1 Mini because I could save myself the step down converter for the ESP (afaik the Wemos D1 accepts 7,4V without any real problems, whereas the C3 Super Mini only accepts 3,3 or 5V).

carlalldis, 2025-05-06:
Must have been a blip for me. I've been testing again and it seems to work 100% every time now. I've put a logging fix PR into @SFrijters repo - thanks!

Rbel12b, 2026-02-01:
> You may want to look at https://github.com/Rbel12b/legoino or my own fork https://github.com/SFrijters/legoino . I haven't had time to test it on hardware yet, but the compilation test suite passes.

You may not want to use [my fork](https://github.com/Rbel12b/Legoino) since it no longer seems to work, and i won't update it, I'm working on a completly new library for Lpf2 Hubs/Devices: [Rbel12b/Lpf2](https://github.com/Rbel12b/Lpf2), but it's currently work-in-progress